volkszaehler/libsml #6: do not printf errors on stdout and add "libsml - " to error message

devZer0, 2015-01-07:
please consider using fprintf(stderr,"libsml - error:.....\n",value); or similar instead

3 reasons for that:

1st:
root@wheezy:~/libsml# while true;do cat /dev/ttyUSB0 |./sml_server_stdio ;done
error: unknown type in sml_value_to_double
error: unknown type in sml_value_to_double
14797050.70 Wh
2012.40 Wh
error: unknown type in sml_value_to_double
20.70 W
^C

where is this error coming from?
how do i get rid of those messages when i want to parse the data ?

if being printed to stderr, you can simple redirect the error messages to /dev/null or to an error-logfile:

root@wheezy:~/libsml# while true;do cat /dev/ttyUSB0 |./sml_server_stdio ;done 2>/dev/null
14797050.80 Wh
2012.40 Wh
21.10 W
^C

2nd:
Because of this behaviour, eople already re-implement sml_value_to_double() within their code - see: http://knx-user-forum.de/201949-post60.html

3rd:
it`s more clear then, that the error message are from libsml and not from the program implementing the library if you add libsml to the message

Note from r00t-:
more correctly, a library should ideally not print errors itself at all,
but report them to the caller.
(for example through a configurable error logging callback.)

yes, i second that - but that`s a little bit more work

regards
roland

root@wheezy:~/libsml/libsml/sml/src# grep print * |egrep "warning|error"
sml_file.c: printf("warning: could not read the whole file\n");
sml_message.c: printf("error: message type %04X not yet implemented\n", *(msg_body->tag));
sml_message.c: printf("error: message type %04X not yet implemented\n", *(message_body->tag));
sml_transport.c: printf("error: unrecognized sequence\n");
sml_tree.c: printf("error: unknown tag in %s\n", FUNCTION);
sml_value.c: printf("error: unknown type in %s\n", FUNCTION);

r00t-, 2015-01-07:
so, why no pull request for that yet?
will you make a commit anytime soon, or should one of us have a look?

devZer0, 2015-01-07:
i would first like to hear what others think about this proposal

r00t-, 2015-01-07:
it's an ovious improvement, just do it.
the callback version would be nicer, but is more work, might do that later.

andig, 2015-02-28:
Applied https://github.com/volkszaehler/libsml/issues/12

closed 2015-02-28 by andig