volkszaehler/libsml #7: no case 0x55 in sml_value_to_double() ?

devZer0, 2015-01-07:
thread: http://volkszaehler.org/pipermail/volkszaehler-dev/2015-January/004049.html

is this possibly a bug , that libsml does not seem to handle value->type 0x55 ? 

from the mail conversation:

according to 
https://www.bsi.bund.de/SharedDocs/Downloads/DE/BSI/Publikationen/TechnischeRichtlinien/TR03109/TR-03109-1_Anlage_Feinspezifikation_Drahtgebundene_LMN-Schnittstelle_Teilb.pdf
and
http://wiki.volkszaehler.org/hardware/channels/meters/power/edl-ehz/emh-ehz-h1
 
the value to interpret after "0x55" is "Integer32"
 
77070100100700FF 01 01 621B 52FF 55 FFFFFF47  0177078181C78
 
let`s convert that to  int32 :
 
http://www.binaryconvert.com/result_signed_int.html?hexadecimal=FFFFFF47  (netter converter, btw)
 
result:
-185 !
 
Apparently, your parser does not work correctly (note: the reporter does not use libslm)
 
the following makes me curious, as there is no "case" 0x55    return ...int32   ?
Shouldn`t it return int32 for 0x55, too?

https://github.com/dailab/libsml/blob/master/sml/src/sml_value.c
 
double sml_value_to_double(sml_value *value) {
switch (value->type) {
case 0x51: return *value->data.int8; break;
case 0x52: return *value->data.int16; break;
case 0x54: return *value->data.int32; break;
case 0x58: return *value->data.int64; break;
case 0x61: return *value->data.uint8; break;
case 0x62: return *value->data.uint16; break;
case 0x64: return *value->data.uint32; break;
case 0x68: return *value->data.uint64; break;

r00t-, 2015-01-08:
hier waehre schoen:
- einfach mal ein patch/commit/pull-request
- beispielfile(s) und/fuer tests

andig, 2015-01-08:
> einfach mal ein patch

+1

Ist es nur die Zeile?

    case 0x55: return *value->data.int32; break;

devZer0, 2015-01-08:
kann man sicher problemlos hinzufügen, macht ja nix kaputt - frage ist, ob 0x54 ein typo ist, denn lt. spec gibt es diesen type scheinbar nicht. schlage daher vor 0x55 hinzuzufügen und 0x54 sicherheitshalber drinzulassen - und wenn die von mir vorgeschlagene sammlung von "real world sml daten" mal etwas umfangreicher gewachsen ist (fange jetzt damit an), dann kann man 0x54 ja kicken...

andig, 2015-01-09:
Dann lieber gleich raus wenn kein Grund erkennbar ist und die Spec das nicht hergibt. Machst Du einen PR?

devZer0, 2015-01-09:
werd mich heut abend mit github beschäftigen und kann das dann machen

andig, 2015-01-09:
Lt. Spec existiert auch `0x51` nicht und `0x52` sollte eigentlich 16bit sein. Mir kommt das ganze sehr seltsam vor. Kann es wirklich das der Parser *so* falsch ist und dennoch bis heute ausreichend gut funktioniert?

Also wenn dann komplett gemäß Spec fixen und dann testen...

devZer0, 2015-01-11:
also ich hab mir das jetzt mal ne stunde angeguckt - und verstehs nicht. ich find da auch nix mit offset im code.

wenn ich das gemäss der spec fixe funktioniert die konvertierung nicht mehr - meine beiden EMH zähler spucken nämlich werte von datentypen aus die es gemäss spezifikation garnicht geben dürfte. 

das heisst, fixen gemäss der spec ist NoGo.

Schau übrigens mal hier:
http://code.metager.de/source/xref/wireshark/epan/dissectors/packet-sml.c

da weiss wer was was wir nicht wissen - oder der hatte ne spec, die wir nicht haben. 

ich kontaktier den mal den author....

devZer0, 2015-01-11:
kaum ist man mal ausgeschlafen und geht frisch ans werk, schon sieht man die bäume im wald gleich viel besser ;)

kurzum - hab was gefunden. 

zum zeitpunkt der case/switch routine enthält value->type bereits garnicht mehr den originären wert aus dem telegram, sondern einen wert der mind schon durch eine logische ver-undung und ver-oderung modifiziert wurde - hier mal die relevanten codestellen:

sml_shared.h:#define SML_TYPE_FIELD                             0x70

int sml_buf_get_next_type(sml_buffer *buf) {
        return (buf->buffer[buf->cursor] & SML_TYPE_FIELD);     <--!!!
}

 int type = sml_buf_get_next_type(buf);

 value->type = type;

       switch (type) {
                case SML_TYPE_OCTET_STRING:
                        value->data.bytes = sml_octet_string_parse(buf);
                        break;
                case SML_TYPE_BOOLEAN:
                        value->data.boolean = sml_boolean_parse(buf);
                        break;
                case SML_TYPE_UNSIGNED:
                case SML_TYPE_INTEGER:
                        // get maximal size, if not all bytes are used (example: only 6 bytes for a u64)
                        while (max < ((byte & SML_LENGTH_FIELD) - 1)) {
                                max <<= 1;
                        }

                        value->data.uint8 = sml_number_parse(buf, type, max);
                        value->type |= max;  <-- !!!
                        break;
                default:
                        buf->error = 1;
                        break;
        }

verstehen tu ich allerdings nicht , was das soll

andig, 2015-02-28:
Das hat sich wohl erledigt.

closed 2015-02-28 by andig