volkszaehler/libsml #10: several new warnings after the last commits

devZer0, 2015-01-09:
huh, think you are very productive with merging  :)

since there have so many changes i wanted to give the latest version a try and now getting lots of warnings which have not been there before: 

make[1]: Entering directory '/root/libsml/libsml/sml'
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_file.o src/sml_file.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_attention_response.o src/sml_attention_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_transport.o src/sml_transport.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_octet_string.o src/sml_octet_string.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_shared.o src/sml_shared.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_number.o src/sml_number.c
src/sml_number.c: In function ‘sml_number_init’:
src/sml_number.c:30:49: warning: unused parameter ‘type’ [-Wunused-parameter]
 void *sml_number_init(u64 number, unsigned char type, int size) {
                                                 ^
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_message.o src/sml_message.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_time.o src/sml_time.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_list.o src/sml_list.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_status.o src/sml_status.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_value.o src/sml_value.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_tree.o src/sml_tree.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_boolean.o src/sml_boolean.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_crc16.o src/sml_crc16.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_open_request.o src/sml_open_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_open_response.o src/sml_open_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_list_request.o src/sml_get_list_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_list_response.o src/sml_get_list_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_close_request.o src/sml_close_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_close_response.o src/sml_close_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_set_proc_parameter_request.o src/sml_set_proc_parameter_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_proc_parameter_response.o src/sml_get_proc_parameter_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_pack_response.o src/sml_get_profile_pack_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_list_request.o src/sml_get_profile_list_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_list_response.o src/sml_get_profile_list_response.c
ld -luuid -shared -soname libsml.so.1 -o lib/libsml.so.1 src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src/sml_open_request.o src/sml_open_response.o src/sml_get_list_request.o src/sml_get_list_response.o src/sml_close_request.o src/sml_close_response.o src/sml_set_proc_parameter_request.o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_response.o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_response.o src/sml_get_profile_list_request.o src/sml_get_profile_list_response.o
ar -rs lib/libsml.a src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src/sml_open_request.o src/sml_open_response.o src/sml_get_list_request.o src/sml_get_list_response.o src/sml_close_request.o src/sml_close_response.o src/sml_set_proc_parameter_request.o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_response.o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_response.o src/sml_get_profile_list_request.o src/sml_get_profile_list_response.o
ar: creating lib/libsml.a
ld -r -o lib/libsml.o src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src/sml_open_request.o src/sml_open_response.o src/sml_get_list_request.o src/sml_get_list_response.o src/sml_close_request.o src/sml_close_response.o src/sml_set_proc_parameter_request.o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_response.o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_response.o src/sml_get_profile_list_request.o src/sml_get_profile_list_response.o
make[1]: Leaving directory '/root/libsml/libsml/sml'
make[1]: Entering directory '/root/libsml/libsml/examples'
cc -I../sml/include/ -g -std=c99 -Wall -Wextra -pedantic -c sml_server.c -o sml_server.o
sml_server.c: In function ‘main’:
sml_server.c:79:14: warning: unused parameter ‘argc’ [-Wunused-parameter]
 int main(int argc, char **argv) {
              ^
sml_server.c:79:27: warning: unused parameter ‘argv’ [-Wunused-parameter]
 int main(int argc, char **argv) {
                           ^
cc -I../sml/include/ -g -std=c99 -Wall -Wextra -pedantic sml_server.o -luuid ../sml/lib/libsml.a -o sml_server -luuid
make[1]: Leaving directory '/root/libsml/libsml/examples'
make[1]: Entering directory '/root/libsml/libsml/test'
make[2]: Entering directory '/root/libsml/libsml/sml'
make[2]: Nothing to be done for 'libsml'.
make[2]: Leaving directory '/root/libsml/libsml/sml'
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c unity/unity.c -o unity/unity.o
unity/unity.c:17:8: warning: missing braces around initializer [-Wmissing-braces]
 struct _Unity Unity = { 0 , 0 , 0 , 0 , 0 , 0 , 0 , 0 , { 0 } };
        ^
unity/unity.c:17:8: warning: (near initialization for ‘Unity.AbortFrame[0]’) [-Wmissing-braces]
unity/unity.c:17:8: warning: missing initializer for field ‘__mask_was_saved’ of ‘struct __jmp_buf_tag’ [-Wmissing-field-initializers]
In file included from unity/unity_internals.h:11:0,
                 from unity/unity.h:12,
                 from unity/unity.c:7:
/usr/include/setjmp.h:41:9: note: ‘__mask_was_saved’ declared here
     int __mask_was_saved; /* Saved the signal mask?  */
         ^
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c unity/unity_fixture.c -o unity/unity_fixture.o
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c src/test_helper.c -o src/test_helper.o
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c src/sml_octet_string_test.c -o src/sml_octet_string_test.o
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c src/sml_buffer_test.c -o src/sml_buffer_test.o
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c src/sml_number_test.c -o src/sml_number_test.o
[... cut ...]
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -luuid unity/unity.o unity/unity_fixture.o src/test_helper.o src/sml_octet_string_test.o src/sml_buffer_test.o src/sml_number_test.o src/sml_boolean_test.o src/sml_value_test.o src/sml_status_test.o src/sml_list_test.o src/sml_time_test.o src/sml_tree_test.o src/sml_file_test.o src/sml_open_request_test.o src/sml_get_profile_pack_request_test.o src/sml_message_test.o ../sml/lib/libsml.a test_main.c -o test -luuid
Unity test run 1 of 1
......................................................................................
-----------------------
86 Tests 0 Failures 0 Ignored
OK
make[1]: Leaving directory '/root/libsml/libsml/test'

devZer0, 2015-01-09:
compressed view :

src/sml_number.c:30:49: warning: unused parameter ‘type’ [-Wunused-parameter]
void sml_number_init(u64 number, unsigned char type, int size) 

sml_server.c:79:14: warning: unused parameter ‘argc’ [-Wunused-parameter]
int main(int argc, char *argv) {
^
sml_server.c:79:27: warning: unused parameter ‘argv’ [-Wunused-parameter]
int main(int argc, char *argv) {
^

unity/unity.c:17:8: warning: missing braces around initializer [-Wmissing-braces]
struct _Unity Unity = { 0 , 0 , 0 , 0 , 0 , 0 , 0 , 0 , { 0 } };
^
unity/unity.c:17:8: warning: (near initialization for ‘Unity.AbortFrame[0]’) [-Wmissing-braces]
unity/unity.c:17:8: warning: missing initializer for field ‘__mask_was_saved’ of ‘struct __jmp_buf_tag’ [-Wmissing-field-initializers]
In file included from unity/unity_internals.h:11:0,
from unity/unity.h:12,
from unity/unity.c:7:
/usr/include/setjmp.h:41:9: note: ‘__mask_was_saved’ declared here
int __mask_was_saved; / Saved the signal mask? */

r00t-, 2015-01-09:
- firs two are duplicate of https://github.com/volkszaehler/libsml/issues/9
- unity: i guess we should update this? or make it an external dependancy?