volkszaehler/libsml #19: Build warning on raspbian

andig, 2016-05-13:
make[2]: Leaving directory '/home/pi/vzlogger/libs/libsml/sml'
	cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c unity/unity.c -o unity/unity.o
	unity/unity.c:17:8: warning: missing braces around initializer [-Wmissing-braces]
	 struct _Unity Unity = { 0 , 0 , 0 , 0 , 0 , 0 , 0 , 0 , { 0 } };
		^
	unity/unity.c:17:8: warning: (near initialization for ‘Unity.AbortFrame[0]’) [-Wmissing-braces]
	unity/unity.c:17:8: warning: missing initializer for field ‘__mask_was_saved’ of ‘struct __jmp_buf_tag’ [-Wmissing-field-initializers]
	In file included from unity/unity_internals.h:11:0,
			 from unity/unity.h:12,
			 from unity/unity.c:7:
	/usr/include/setjmp.h:41:9: note: ‘__mask_was_saved’ declared here
	     int __mask_was_saved; /* Saved the signal mask?  */
		 ^

hmueller01, 2018-01-13:
@andig could you verify that my change fixed the warning? Maybe someone else on a non Raspberry system could also verify this. Thanks.

hmueller01, 2018-01-13:
Ok, my implementation breaks compiling on e.g. OSX (x86 system).
```
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c unity/unity.c -o unity/unity.o
unity/unity.c:19:22: error: designator in initializer for scalar type 'int'
    .AbortFrame = { {.__jmpbuf = {0}, .__mask_was_saved = 0, .__saved_mask = {{0}} } }
                     ^~~~~~~~~~~~~~~
1 error generated.
make[1]: *** [unity/unity.o] Error 1
```
It looks like `jmp_buf` (the type of `AbortFrame`) is defined differently depending on the environment.

RaspberryPi defines it like this
```
/* Calling environment, plus possibly a saved signal mask.  */
struct __jmp_buf_tag
  {
    /* NOTE: The machine-dependent definitions of `__sigsetjmp'
       assume that a `jmp_buf' begins with a `__jmp_buf' and that
       `__mask_was_saved' follows it.  Do not move these members
       or add others before it.  */
    __jmp_buf __jmpbuf;         /* Calling environment.  */
    int __mask_was_saved;       /* Saved the signal mask?  */
    __sigset_t __saved_mask;    /* Saved signal mask.  */
  };
typedef struct __jmp_buf_tag jmp_buf[1];

```
OSX like this
```
typedef int jmp_buf[_JBLEN];
```

One solution might be a define where we can select a special case for the Raspberry (or other Linux systems?). But I haven't found one. Currently I have no idea how to solve this.

Can someone please undo my change. Tx.

devZer0, 2019-04-11:
on my raspi with raspian/debian 9.8  it builds fine and works fine

```
root@pivpn2:/tmp/libsml# make
make[1]: Entering directory '/tmp/libsml/sml'
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_file.o src/sml_file.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_attention_response.o src/sml_attention_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_transport.o src/sml_transport.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_octet_string.o src/sml_octet_string.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_shared.o src/sml_shared.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_number.o src/sml_number.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_message.o src/sml_message.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_time.o src/sml_time.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_list.o src/sml_list.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_status.o src/sml_status.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_value.o src/sml_value.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_tree.o src/sml_tree.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_boolean.o src/sml_boolean.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_crc16.o src/sml_crc16.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_open_request.o src/sml_open_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_open_response.o src/sml_open_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_list_request.o src/sml_get_list_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_list_response.o src/sml_get_list_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_close_request.o src/sml_close_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_close_response.o src/sml_close_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_set_proc_parameter_request.o src/sml_set_proc_parameter_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_proc_parameter_response.o src/sml_get_proc_parameter_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_pack_response.o src/sml_get_profile_pack_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_list_request.o src/sml_get_profile_list_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_list_response.o src/sml_get_profile_list_response.c
ld -luuid -shared -soname libsml.so.1 -o lib/libsml.so.1 src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src/sml_open_request.o src/sml_open_response.o src/sml_get_list_request.o src/sml_get_list_response.o src/sml_close_request.o src/sml_close_response.o src/sml_set_proc_parameter_request.o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_response.o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_response.o src/sml_get_profile_list_request.o src/sml_get_profile_list_response.o
ar -rs lib/libsml.a src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src/sml_open_request.o src/sml_open_response.o src/sml_get_list_request.o src/sml_get_list_response.o src/sml_close_request.o src/sml_close_response.o src/sml_set_proc_parameter_request.o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_response.o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_response.o src/sml_get_profile_list_request.o src/sml_get_profile_list_response.o
ar: creating lib/libsml.a
ld -r -o lib/libsml.o src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src [... cut ...]
.......................................................................................
-----------------------
87 Tests 0 Failures 0 Ignored
OK
make[1]: Leaving directory '/tmp/libsml/test'

root@pivpn2:/tmp/libsml-testing#  cat EMH_eHZ-GW8E2A500AK2.bin |  /tmp/libsml/examples/sml_server - | head -n10
129-129:199.130.3*255#EMH#
1-0:0.0.0*255#02280816#
1-0:1.8.1*255#14798112.9#Wh
1-0:1.8.2*255#2012.4#Wh
0-0:96.1.255*255#0002280816#
1-0:1.7.0*255#13.8#W
129-129:199.130.3*255#EMH#
1-0:0.0.0*255#02280816#
1-0:1.8.1*255#14798113.0#Wh
1-0:1.8.2*255#2012.4#Wh
```

devZer0, 2019-04-14:
i think this can be closed

devZer0, 2020-02-20:
also builds fine on debian 9.4 on raspberry, could you please close this @andig ?

also have running some influx/graphana setup with sml_server for weeks now @dingfabrik/colgone for weeks via raspberry as "bridge" between eHZ and influxdb without a problem

closed 2020-02-21 by andig

r00t-, 2021-03-25:
when this was opened, our copy of unity was already three years old,
we should probably simply have upgraded it, see #99