volkszaehler/libsml #25: libsml: error: unknown type 0 in sml_value_to_double

devZer0, 2018-01-15:
i'm getting this libsml error messages. think they have been there for a long time, but i'd like to get rid of those and i think libsml does parsing wrong or incomplete.

Dr.Neuhaus Smarty ix-130:

./sml_server /dev/ttyUSB0 
SML file (3 SML messages, 308 bytes)
SML message  101
SML message  701
SML message  201
OBIS data
libsml: error: unknown type 0 in sml_value_to_double
libsml: error: unknown type 0 in sml_value_to_double
1-0:2.8.0*255#4373.0#Wh
1-0:2.8.1*255#4373.0#Wh
1-0:15.7.0*255#60.0#W
libsml: error: unknown type 0 in sml_value_to_double

and nearly the same (mind the different position) with EMH eHZ-GW8E2A500AK2

./sml_server /dev/ttyUSB0 
SML file (3 SML messages, 236 bytes)
SML message  101
SML message  701
SML message  201
OBIS data
libsml: error: unknown type 0 in sml_value_to_double
libsml: error: unknown type 0 in sml_value_to_double
1-0:1.8.1*255#14798741.6#Wh
1-0:1.8.2*255#2012.4#Wh
libsml: error: unknown type 0 in sml_value_to_double
1-0:1.7.0*255#24.9#W

smartmeter-obis (nodejs lib) prints this ones, so apparently libsml is not able to decode some of the obis values:

cat /dev/ttyUSB0 |node StdInSml.js 
129-129:199.130.3*255: Manufacturer ID = DNT
1-0:0.0.9*255: Device ID = 0901444e540100002030
1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 4.373 kWh
1-0:2.8.1*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T1) = 4.373 kWh
1-0:2.8.2*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (T2) = 0 kWh
1-0:15.7.0*255: Momentanwert (Total) = 26 W
129-129:199.130.5*255: Forename/Public Key = fea5d6bc658d734673f5790051d61c2ebd55344a6285686de3b645ee0104d65f5abba6cb68f1bfcb7fcf94f99cd4d845

cat /dev/ttyUSB0 |node StdInSml.js 
129-129:199.130.3*255: Manufacturer ID = EMH
1-0:0.0.0*255: Meter owner number = 02280816
1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 14798.7427 kWh
1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 2.0124 kWh
0-0:96.1.255*255: Factory number = 0002280816
1-0:1.7.0*255: Momentanwert Summe Wirkarbeit Bezug + (Total) = 23.9 W

hmueller01, 2018-01-15:
sml_server currently supports only double values (type 0x51 ... and 0x61 ..., see sml_value_to_double()). If there is another type (e.g. 0 for octet_string) this fails and an error is prompted. You can remove these errors using
`./sml_server /dev/ttyUSB0 2>/dev/null`
But you are right. This is far from perfect. If someone has time to implement more types feel free.

andig, 2018-01-15:
Could you share the actual message that cause the problem to see for which ones decoding is missing?

devZer0, 2018-01-15:
i have added DrNeuhaus eHz here https://github.com/devZer0/libsml-testing

compare the output from smartmeter-obis, think there you also can see what's missing and causing the error

hmueller01, 2018-01-17:
@devZer0 Ok, I implemented something. Could you please test my branch
[https://github.com/hmueller01/libsml/tree/sml-server-mods](https://github.com/hmueller01/libsml/tree/sml-server-mods). It works fine with your dump.

devZer0, 2018-01-17:
cool. trying it. 

apparently i'm getting the following compile error:
make[1]: Verzeichnis »/tmp/sml-hmueller/libsml/examples« wird betreten
cc -I../sml/include/ -g -std=c99 -Wall -Wextra -pedantic -c sml_server.c -o sml_server.o
cc -I../sml/include/ -g -std=c99 -Wall -Wextra -pedantic sml_server.o -lm -luuid ../sml/lib/libsml.a -o sml_server
../sml/lib/libsml.a(sml_octet_string.o): In Funktion `sml_octet_string_generate_uuid':
/tmp/sml-hmueller/libsml/sml/src/sml_octet_string.c:108: Nicht definierter Verweis auf `uuid_generate'
collect2: error: ld returned 1 exit status
make[1]: *** [sml_server] Fehler 1
make[1]: Verzeichnis »/tmp/sml-hmueller/libsml/examples« wird verlassen
make: *** [all] Fehler 2

devZer0, 2018-01-17:
nice, errors are gone , looks a little bit more like smartmeter-obis output now:

# ./sml_server /dev/ttyUSB0 
SML file (3 SML messages, 308 bytes)
SML message  101
SML message  701
SML message  201
OBIS data
129-129:199.130.3*255#DNT#
1-0:0.0.9*255#09 01 44 4e 54 01 00 00 20 30 #
1-0:2.8.0*255#4373.0#Wh
1-0:2.8.1*255#4373.0#Wh
1-0:2.8.2*255#0.0#Wh
1-0:15.7.0*255#0.0#W
129-129:199.130.5*255#fe a5 d6 bc 65 8d 73 46 73 f5 79 00 51 d6 1c 2e bd 55 34 4a 62 85 68 6d e3 b6 45 ee 01 04 d6 5f 5a bb a6 cb 68 f1 bf cb 7f cf 94 f9 9c d4 d8 45 #

hmueller01, 2018-01-18:
@devZer0 can you explain why you had an compile error in the fist place?

devZer0, 2018-01-18:
i still having it, i worked around that in the code by editing the ifndef statement

devZer0, 2018-01-18:
also happens with master, iirc it did not happen some days ago, so must be introduced by a recent change

hmueller01, 2018-01-18:
@Ampelbein changed something in the build system pull #20 that is related with uuid. Can you please make a new issue to address this. Thanks. I will make a pull request to add this change to master, so we can close this.

devZer0, 2018-01-18:
i could fix the issue by exchanging position of $(LIBS) $(LIBSML) in example/Makefile:

before:
sml_server: $(OBJS) $(LIBSML)
	$(CC) $(CFLAGS) $(OBJS) $(LIBS) $(LIBSML) -o sml_server

after:
sml_server: $(OBJS) $(LIBSML)
	$(CC) $(CFLAGS) $(OBJS) $(LIBSML) $(LIBS) -o sml_server

hmueller01, 2018-01-19:
@devZer0 This is fixed, can you please close it. Thanks.

andig, 2018-01-19:
Closed via #31

closed 2018-01-19 by andig