volkszaehler/libsml #26: feature request: please add stdin  and output formatting options

devZer0, 2018-01-15:
i would find

cat /dev/ttyUSB0 >/tmp/binary-sml.dump

cat /tmp/binary-sml.dump | ./sml_server -i stdin 

very useful

this way we would not even need to modify sml_server for proper serial initialization, we could do that outside of sml_server and then use

cat /dev/ttyUSB0 | ./sml_server -i stdin

What i would also find usefule besides vzlogger would be a simple and universal binary which gets sml binary stream as input and does decoded, machine-parseable sml/obis as output.

format1:
json
see smartmeter-obis

format2:
parseable envvars (can be read via pipe, so you can use the power of unix...) 

example:
cat /dev/ttyUSB0 | ./sml_server -i stdin -o envvars
ENVOBISCODE1="1-0:2.8.0*255" ENVOBSINAME1="Zählerstand 1 Summe Wirkarbeit Abgabe - (Total)" ENVOBISVAL="4.373" ENVOBISUNIT="kWh" ENVOBISCODE2="....." ENVOBISNAME2="..."

This way we could simply do:
cat /dev/ttyUSB0 | ./sml_server -o stdin -o envvars | while read line;do eval $line; echo $ENVOBISVAL1;rrdtool $ENVOBISVAL1 ; curl http://server/path?var=$ENVOBISVAL1 ; your_favourite_tool_here $ENVOBISVAL1;... ;done

format3: 
maybe csv or something like that?

r00t-, 2018-01-19:
stdin is here: https://github.com/volkszaehler/libsml/pull/34/commits/133cb99844c44a036b5c01dcb47a18d4f2e7cce8
part of https://github.com/volkszaehler/libsml/pull/34

devZer0, 2018-01-19:
i was surprised yesterday when i found how easy that is to implement. thanks!

closed 2018-01-19 by devZer0