volkszaehler/libsml #29: Compile error "undefined reference to `test_helper_c2ptoi'"

msebald, 2018-01-17:
Copy-pasted from the old repository: https://github.com/dailab/libsml/issues/13

I tried to compile libsml on a fresh Debian 9 Stretch amd64 box. uuid-dev and uuid-runtime is installed. This happens:

```
make[1]: Entering directory '/home/roi/src/libsml/sml'
make[1]: Nothing to be done for 'libsml'.
make[1]: Leaving directory '/home/roi/src/libsml/sml'
make[1]: Entering directory '/home/roi/src/libsml/examples'
make[1]: 'sml_server' is up to date.
make[1]: Leaving directory '/home/roi/src/libsml/examples'
make[1]: Entering directory '/home/roi/src/libsml/test'
make[2]: Entering directory '/home/roi/src/libsml/sml'
make[2]: Nothing to be done for 'libsml'.
make[2]: Leaving directory '/home/roi/src/libsml/sml'
cc -I../sml/include/ -Wall -luuid unity/unity.o unity/unity_fixture.o src/test_helper.o src/sml_octet_string_test.o src/sml_buffer_test.o src/sml_number_test.o src/sml_boolean_test.o src/sml_value_test.o src/sml_status_test.o src/sml_list_test.o src/sml_time_test.o src/sml_tree_test.o src/sml_file_test.o src/sml_open_request_test.o src/sml_get_profile_pack_request_test.o src/sml_message_test.o ../sml/lib/libsml.a test_main.c -o test
src/test_helper.o: In function `hex2binary':
test_helper.c:(.text+0xa3): undefined reference to `test_helper_c2ptoi'
collect2: error: ld returned 1 exit status
Makefile:33: recipe for target 'test' failed
make[1]: *** [test] Error 1
make[1]: Leaving directory '/home/roi/src/libsml/test'
Makefile:3: recipe for target 'all' failed
make: *** [all] Error 2
```

Bug or did I do something wrong?

hmueller01, 2018-01-17:
This is strange. test_helper_c2ptoi is defined in test_helper.c itself. Compiles fine on Raspberry and OSX. I have seen a pull on dailab from @tomweber having compile problems too. He removed the inline keyword from test_helper_c2ptoi definition. Can you please try this.

hmueller01, 2018-01-19:
@andig Can reproduce this with Travis CI?

andig, 2018-01-19:
> Can reproduce this with Travis CI?

No, I can't.

hmueller01, 2018-01-19:
I mean set up a Debian 9 Stretch environment and compile with this.

hmueller01, 2018-01-19:
We changed a lot i the mean time. So on the long run it would make sense to switch over.
Do you have the `static inline uint8_t test_helper_c2ptoi()` definition in your `test/src/test_helper.c` or without `static`? @TheCount changed this in 2013 ...

andig, 2018-01-19:
I'm closing this issue as not related to our repo/software. Feel free to reopen after testing.

closed 2018-01-19 by andig

msebald, 2018-01-19:
It works using this repository. Thank you!

See [here](https://community.openhab.org/t/using-a-power-meter-sml-with-openhab/21923) and [here](https://git.hot-chilli.net/msebald/openhab/src/branch/master/sml) why and what I modified.

hmueller01, 2018-01-19:
@msebald Why do you need a modification anyway? Is it only to exit after the first message?

msebald, 2018-01-19:
@hmueller01 To be honest, I did not check on the code. ;-) I mean checking what exactly is the difference when replacing the complete _transport_receiver_ function. The main reason is that I am not a coder and second I wanted to follow the instructions from the openHAB community 1 by 1 to get it up and running. ;-)

If it is no longer necessary to alter the code or of it is possible to get the needed output with an commandline option it would be a nice feature for all the people using libsml with openHAB.

hmueller01, 2018-01-19:
@msebald This is why I am asking. What do you/openhab need? If you find that out maybe I can help to implement that and we can have it in the libsml repo. Let me know.

msebald, 2018-01-19:
This is what I get when using the modified version:
```
[16:32:18] root@openhabian:/home/roi/volkszaehler/libsml/examples# ./sml_server /dev/sml0
1-0:1.8.0*255#20234.615#
1-0:1.8.1*255#20234.615#
1-0:16.7.0*255#1.081#
```
What is the original output? ;-) I could check, but I am sure that you will know it.

The output is parsed with a openHAB rule, see [here](https://git.hot-chilli.net/msebald/openhab/src/branch/master/sml/sml.rules).

hmueller01, 2018-01-23:
Sorry, I got no message that you answered. I just saw it right now.
Is this the output of your modified sml_server? Does your "server" exit after reading one "record"?

msebald, 2018-01-23:
Yes, that is the output. And yes, sml_server exits after the output. It is started by openHAB every x seconds, according to how you configured the rule. For me it is every 30 seconds.

hmueller01, 2018-01-23:
Is it a problem if all sml messeges (e.g. like finger print) are in the output. If not, we only need a switch from permanent read to single read, right?

msebald, 2018-01-23:
Yes, maybe. I am just a user and a very bad programmer. Did you have a chance to check what is changed for our use case with openHAB?