volkszaehler/libsml #30: libsml rework/todo

devZer0, 2018-01-17:
see https://demo.volkszaehler.org/pipermail/volkszaehler-dev/2015-January/004075.html
and https://marc.info/?l=volkszaehler-dev&m=140118891532418&w=2   (just adding it here so it won`t get lost)

r00t-, 2018-01-19:
ich habe die mails durchgelesen,
aber kann da wenig konkretes herauslesen,
ich waehre dafuer, zu konkreteren einzelpunkten issues anzulegen.

"make install" koennte einer sein.
wobei ich den patch von @peterevertz schon 2015 reviewed und fuer schlecht befunden habe:
https://github.com/peterevertz/libsml/commit/d7ab04aed281e8bfed20f678f39249b1a9a1e213#commitcomment-9256588

hmueller01, 2018-01-19:
I implemented a very basic "make install" which is a little better than Peters. But for sure, it is far from perfect. Should we change e.g. to cmake?

devZer0, 2018-01-19:
ja du hast recht, das ist recht allgemein/unkonkret. aber einiges ist da eher schwer umsetzbar - lieber ein einzelnes unerledigtes erinnerungsticket als mehrere, oder? entscheid du, was wir machen

r00t-, 2018-01-19:
regarding "make install",
please realize that it's rather pointless.
vzlogger is the only user of libsml,
it makes much more sense to just configure it to statically link against the libs from the source tree.
the only reason it was added is because users are used to execute "make" followed by "make install" cargo-cult style.

andig, 2018-01-19:
Unklar was hier gefragt ist- close?

closed 2018-11-21 by andig