volkszaehler/libsml #33: let's build "parsesml" - rename sml_server

devZer0, 2018-01-19:
i never understood why the example prog being called "sml_server". it's not a "server" prog, imho. you cannot connect to that and there is no client.

can't/shouldn't we turn that into a flexible and minimalistic commandline parser utility and call that "parsesml" instead?

r00t-, 2018-01-19:
you could make it a server by running it in inetd ;)

also,  the feature to read from a serial port seems useful, but conflicts slightly with the "parsesml" name?

devZer0, 2018-01-19:
to be honest, do we need to handle/init the serial port stuff in sml_server at all?  to do it right, we need to rework all the initialization stuff for proper init, i.e. need more cmdline opts.  can't we do that much better with the appropriate and specialized cmdline util?  i wonder if we can't drop opening files from sml_server and use stty/setserial instead and then do : sml_server </dev/$SERIAL or cat /dev/$SERIAL | sml_server   

no name conflict with parsesml then, too. the tool simply does one thing: parsing sml and translating that into readable content.

hmueller01, 2018-01-19:
For me it is the "server", as it runs permanent in background and reads the data from the meter. So it "serves" the data ...

Init the serial device is important as default values could be different and your solution would not work that way.

devZer0, 2018-01-19:
yes it IS important, but what do you mean with "default values could be different" ?  is there anything in the init stuff which can't be done with stty? see https://linux.die.net/man/1/stty

hmueller01, 2018-01-19:
No, do not think there is anything. Can be done by stty. If we rename it to parser we have to remove this stuff from the code and need to use stty. This is also only an example, how one could implement something with this lib. So it could also be renamed to sml_example.c ...
If we some day implement your output formatting proposals of #26 we we could start new and call this parsesml, parse_sml or sml_parse and remove in this new code the tty stuff. Maybe you should rename your issue like this and add the proposals as you closed #26. But I like to stay with an example including tty init.

devZer0, 2018-01-19:
ok

andig, 2018-01-22:
Note sure about scope of this discussion. +1 for renaming to sml_parser or parsesml.

closed 2018-01-22 by andig