volkszaehler/libsml #40: OSX build warning

andig, 2018-01-20:
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c unity/unity.c -o unity/unity.o
	cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c unity/unity_fixture.c -o unity/unity_fixture.o
	cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c src/test_helper.c -o src/test_helper.o
	src/test_helper.c:38:23: warning: unused function 'test_helper_c2toi' [-Wunused-function]
	static inline uint8_t test_helper_c2toi(uint8_t c1, uint8_t c2){
	                      ^

with current master

hmueller01, 2018-01-20:
That's correct. The function `test_helper_c2toi` is implemented and not used. The option `-Wall` includes `-Wunused-function`. Why this is not reported under Linux is unknown to me. If this is annoying we could comment out this function. I do not know if this code is helpful in the future ...

devZer0, 2018-01-25:
you may get a clue here:
https://stackoverflow.com/questions/13224209/gcc-wunused-function-not-working-but-other-warnings-are-working

commenting out sounds good to me - whoever wants to use this function can easily comment it in again. and since there isn't an api/function documentation which could alread mention that the lib contains this, it won't be a problem. i also doubt any external project is using this...

closed 2018-11-21 by andig