volkszaehler/libsml #42: sml_example does not honour EOF when using STDIN

devZer0, 2018-01-21:
if you do 

cat binarysml.dump|./sml_server -

the command does never exit.

r00t-, 2018-01-22:
see also https://github.com/volkszaehler/libsml/issues/32#issuecomment-358830097

r00t-, 2018-01-22:
fix goes here:
https://github.com/volkszaehler/libsml/blob/master/sml/src/sml_transport.c#L111

devZer0, 2018-01-22:
did you test that with my binary dumps?  fix does not work for me

andig, 2018-01-22:
>  fix does not work for me

That's why its work in progress.

devZer0, 2018-01-22:
i found http://pubs.opengroup.org/onlinepubs/9699919799/functions/read.html

When attempting to read from an empty pipe or FIFO:
    If no process has the pipe open for writing, read() shall return 0 to indicate end-of-file.

and added check like the following (see below).

that works for me, but i'm not sure if it's the correct way to handle that, i'm an awful bad programmer.

size_t sml_read(int fd, fd_set *set, unsigned char *buffer, size_t len) {

        ssize_t r = 0;
        size_t tr = 0;

        while (tr < len) {
                select(fd + 1, set, 0, 0, 0);
                if (FD_ISSET(fd, set)) {
                        r = read(fd, &(buffer[tr]), len - tr);
                        if (r == 0 )break;
                        if (r < 0 ) continue;

                        tr += r;
                }
        }
        return tr;
}

andig, 2018-11-02:
Fixed by #51

closed 2018-11-02 by andig