volkszaehler/libsml #46: sml_server: add a switch to exit after one packet

hmueller01, 2018-01-23:
Off topic I discussed in #29 with @msebald that sml_server is used by the @openhab project [rule sml](https://git.hot-chilli.net/msebald/openhab/src/branch/master/sml/sml.rules). At the moment sml_server.c is copied there and modified slightly. The main reason for the change is, that the program needs to stop after processing one packet. My idea is to add a switch to do that. Do I get support for this and how should the switch be called? E.g. "-s" for single?

andig, 2018-01-23:
I would have appreciated openhab proposing changes here.

devZer0, 2018-01-23:
smartmeter-obis got such switch just recently. i would recommend everyone developing on libsml should have a look at that lib - just to have something to compare with

msebald, 2018-01-24:
Thank you for the new issue and that you are willing to include the needs of the openHAB people into libsml.

I just checked. In my case the unmodified sml_server produces the following output, which also loops:
```
SML file (3 SML messages, 316 bytes)
SML message  101
SML message  701
SML message  201
OBIS data
129-129:199.130.3*255#EMH#
1-0:0.0.9*255#06 45 4d 48 01 0b c1 a5 e7 2b #
1-0:1.8.0*255#20355093.1#Wh
1-0:1.8.1*255#20355093.1#Wh
1-0:1.8.2*255#0.0#Wh
1-0:16.7.0*255#592.3#W
129-129:199.130.5*255#T04 77 07 9e 79 9f 7c 4d 74 45 e0 b8 6b 6c fd 50 88 8a ab 9a 3a 5b 05 b8 4b fa d3 24 e1 f4 70 d1 30 39 e1 a4 93 23 18 9a 24 ec 3b ec c1 83 15 #
```
The modified sml_server produces the following output and then exists:
```
1-0:1.8.0*255#20234.615#
1-0:1.8.1*255#20234.615#
1-0:16.7.0*255#1.081#
```
So it it not just about a switch to terminate sml_server after a single run.

What I do not know exactly as my knowledge about Java (which is used for openHAB and also the linked rule) is basic: Does the unmodified sml_server output work? I would say no as it looks into all lines and makes something. Sorry that I cannot be more clear on that as I am really not a programmer...

hmueller01, 2018-01-24:
No, it will not work as it is. The rule just takes line 0 and 2, which won't work on all meters today. It needs to parse the lines it reads for 1.8.0 and 16.7.0. This should be quite simple.

hmueller01, 2018-02-02:
I made a branch which includes the -s. Please check, if this works. [sml_server_switch](https://github.com/hmueller01/libsml/tree/sml_server_switch)
Please keep in mind that you still need to add a message parsing in the rule. This can not be done by sml_server.

hmueller01, 2018-02-06:
@msebald Did you have time to test my code?
I tried to change your rule to parse the object names:
```
		// getting the payload of the meter
		var String meter_payload = executeCommandLine("/home/openhabian/volkszaehler/libsml/examples/sml_server -s /dev/ttyUSB0", 5000) // note the argument /dev/ttyUSB0 or /dev/sml0 - your meter-device-id goes here
		// splitting the payload - first in lines, then getting counterStr and consumptionStr with delimiter "#"
		val lines = meter_payload.split('\n')
		for (String line : lines) {
			if (line != null) {
				// split the line separated into object name and value
				var lineStr = line.split('#')
				var obj_name = lineStr.get(0)
				var value = lineStr.get(1)
				if (obj_name == "1-0:1.8.0*255") {
					if (EHZ_status_power.state != value)
						EHZ_status_power.sendCommand(value)
				} else if (obj_name == "1-0:16.7.0*255") {
					EHZ_consumption_power.postUpdate(value)
				}
			}
		}
```
But be aware. I am not familiar in writing Xtend code and have not tested it with openHAB.

msebald, 2018-02-09:
No. But I will. I'm on vacation. Next week I'm back home again.

hmueller01, 2018-08-13:
@msebald Have you had a chance to test the [code](https://github.com/hmueller01/libsml/tree/sml_server_switch) and the modified parsing? Just saw that this issue is still open.

hmueller01, 2018-11-03:
No answer from @msebald. But as I created a pull request, I'll close this.

closed 2018-11-03 by hmueller01