volkszaehler/libsml #50: Process hangs once a week

tobylorenz, 2018-10-30:
I wrote a small program that uses libsml to read meter values and publish them via MQTT. Once a week this process hangs with 100% cpu utilization. I've attached GDB to the process and run a backtrace. Here's the output:
```
root@heizung:/usr/local/bin# gdb sml2mqtt 2190
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
"/usr/local/bin/sml2mqtt": not in executable format: File format not recognized
Attaching to process 2190
[New LWP 2191]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
0xb6f57460 in read () from /lib/arm-linux-gnueabihf/libpthread.so.0
(gdb) info proc all
process 2190
warning: target file /proc/2190/cmdline contained unexpected null characters
cmdline = '/usr/sbin/sml2mqtt'
cwd = '/'
exe = '/usr/sbin/sml2mqtt'
Mapped address spaces:

        Start Addr   End Addr       Size     Offset objfile
        0xb6100000 0xb6121000    0x21000        0x0 
        0xb6121000 0xb6200000    0xdf000        0x0 
        0xb6233000 0xb6234000     0x1000        0x0 
        0xb6234000 0xb6a33000   0x7ff000        0x0 [stack:2191]
        0xb6a33000 0xb6a39000     0x6000        0x0 /lib/arm-linux-gnueabihf/libnss_files-2.24.so
        0xb6a39000 0xb6a49000    0x10000     0x6000 /lib/arm-linux-gnueabihf/libnss_files-2.24.so
        0xb6a49000 0xb6a4a000     0x1000     0x6000 /lib/arm-linux-gnueabihf/libnss_files-2.24.so
        0xb6a4a000 0xb6a4b000     0x1000     0x7000 /lib/arm-linux-gnueabihf/libnss_files-2.24.so
        0xb6a4b000 0xb6a51000     0x6000        0x0 
        0xb6a51000 0xb6a53000     0x2000        0x0 /lib/arm-linux-gnueabihf/libdl-2.24.so
        0xb6a53000 0xb6a62000     0xf000     0x2000 /lib/arm-linux-gnueabihf/libdl-2.24.so
        0xb6a62000 0xb6a63000     0x1000     0x1000 /lib/arm-linux-gnueabihf/libdl-2.24.so
        0xb6a63000 0xb6a64000     0x1000     0x2000 /lib/arm-linux-gnueabihf/libdl-2.24.so
        0xb6a64000 0xb6a6e000     0xa000        0x0 /usr/lib/arm-linux-gnueabihf/libcares.so.2.2.0
        0xb6a6e000 0xb6a7e000    0x10000     0xa000 /usr/lib/arm-linux-gnueabihf/libcares.so.2.2.0
        0xb6a7e000 0xb6a7f000     0x1000     0xa000 /usr/lib/arm-linux-gnueabihf/libcares.so.2.2.0
        0xb6a7f000 0xb6a80000     0x1000     0xb000 /usr/lib/arm-linux-gnueabihf/libcares.so.2.2.0
        0xb6a80000 0xb6bdb000   0x15b000        0x0 /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1
        0xb6bdb000 0xb6bea000     0xf000   0x15b000 /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1
        0xb6bea000 0xb6bf9000     0xf000   0x15a000 /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1
        0xb6bf9000 0xb6c00000     0x7000   0x169000 /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1
        0xb6c00000 0xb6c04000     0x4000        0x0 
        0xb6c04000 0xb6c3c000    0x38000        0x0 /usr/lib/arm-linux-gnueabihf/libssl.so.1.1
        0xb6c3c000 0xb6c4b000     0xf000    0x38000 /usr/lib/arm-linux-gnueabihf/libssl.so.1.1
        0xb6c4b000 0xb6c4e000     0x3000    0x37000 /usr/lib/arm-linux-gnueabihf/libssl.so.1.1
        0xb6c4e000 0xb6c52000     0x4000    0x3a000 /usr/lib/arm-linux-gnueabihf/libssl.so.1.1
        0xb6c52000 0xb6c57000     0x5000        0x0 /lib/arm-linux-gnueabihf/librt-2.24.so
        0xb6c57000 0xb6c66000     0xf000     0x5000 /lib/arm-linux-gnueabihf/librt-2.24.so
        0xb6c66000 0xb6c67000     0x1000     0x4000 /lib/arm-linux-gnueabihf/librt-2.24.so
        0xb6c67000 0xb6c68000     0x1000     0x5000 /lib/arm-linux-gnueabihf/librt-2.24.so
        0xb6c68000 0xb6c6b000     0x3000        0x0 /lib/arm-linux-gnueabihf/libuuid.so.1.3.0
        0xb6c6b000 0xb6c7a000     0xf000     0x3000 /lib/arm-linux-gnueabihf/libuuid.so.1.3.0
        0xb6c7a000 0xb6c7b000     0x1000     0x2000 /lib/arm-linux-gnueabihf/libuuid.so.1.3.0
        0xb6c7b000 0xb6c7c000     0x1000     0x3000 /lib/arm-linux-gnueabihf/libuuid.so.1.3.0
---Type <return> to continue, or q <return> to quit--- 
        0xb6c7c000 0xb6c85000     0x9000        0x0 /usr/lib/arm-linux-gnueabihf/libmosquitto.so.1
        0xb6c85000 0xb6c94000     0xf000     0x9000 /usr/lib/arm-linux-gnueabihf/libmosquitto.so.1
        0xb6c94000 0xb6c95000     0x1000     0x8000 /usr/lib/arm-linux-gnueabihf/libmosquitto.so.1
[... 100 lines cut ...]
jiffies remaining in current time slice: 20
'nice' value: 0
jiffies until next timeout: 2
jiffies until next SIGALRM: 0
start time (jiffies since system boot): 901087
Virtual memory size: 15482880
Resident set size: 604
rlim: 4294967295
Start of text: 0xb6fba000
End of text: 0xb6fc1048
Start of stack: 0xbe90bdf0
(gdb) info threads
  Id   Target Id         Frame 
* 1    Thread 0xb6fb1800 (LWP 2190) "sml2mqtt" 0xb6f57460 in read () from /lib/arm-linux-gnueabihf/libpthread.so.0
  2    Thread 0xb6a32420 (LWP 2191) "sml2mqtt" 0xb6cac6f4 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) bt
#0  0xb6f57460 in read () from /lib/arm-linux-gnueabihf/libpthread.so.0
#1  0xb6f3451e in sml_read (fd=6, set=0xbe909ad4, buffer=0xbe907c28 "-", len=4) at src/sml_transport.c:45
#2  0xb6f34698 in sml_transport_read (fd=6, buffer=0xbe909b94 "\033\033\033\033\001\001\001\001v\005\n\035\317\242b", max_len=8096) at src/sml_transport.c:79
#3  0xb6f34784 in sml_transport_listen (fd=6, 
    sml_transport_receiver=0xb6fbe5d1 <SML::transport_listen()::{lambda(unsigned char*, unsigned int)#1}::_FUN(unsigned char*, unsigned int)>) at src/sml_transport.c:112
#4  0xb6fbc5ea in main ()
```

It seems to hang in sml_transport_listen. Any ideas on how to solve this?

ththvseo, 2018-10-30:
i'm having the suspicion that it's caused by read() failing, and the code not detecting that, and retrying forever.    
https://github.com/volkszaehler/libsml/blob/master/sml/src/sml_transport.c#L46

can you put a breakpoint in sml_read() and single-step through it (when the error has already occured)?

can you also try to attach to the still working process and just
```call close(6)``` from gdb, to see if that reproduces the issue?

ththvseo, 2018-10-30:
heck, we even have a (stalled) merge-reuqest already, that's why it seems so familiar:
https://github.com/volkszaehler/libsml/pull/43

tobylorenz, 2018-10-30:
I'll do, when it hangs the next time. ;-)

r00t-, 2018-10-30:
(deleted my comment above and re-posting/wording it:)

this seems to be an instance of libsml ignoring errors and EOF from read(), and retrying forever.

the code in question:
https://github.com/volkszaehler/libsml/blob/master/sml/src/sml_transport.c#L46

we even have a (stalled) merge-request already: #43

if you you put a breakpoint in sml_read() and single-step through it (when the error has already occured), you should see what is happening.

alternatively, without waiting for the error to repeat,
can you also try to attach to the still working process and just
`call close(6)` from gdb, and that should reproduce the issue.

tobylorenz, 2018-10-31:
You are right, when I `call close(6)`, the process goes again into 100% cpu load. When I reattach to the process and single step then it shows
```
Attaching to process 7132
[New LWP 7133]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
0xb6d59dc4 in select () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) break sml_read
Breakpoint 1 at 0xb6f684ba: file src/sml_transport.c, line 38.
(gdb) step 
Single stepping until exit from function select,
which has no line number information.
0xb6d675e8 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) step
Cannot find bounds of current function
```

I'll try the same with the version in https://github.com/volkszaehler/libsml/pull/51.

tobylorenz, 2018-10-31:
I've tested the version in #51. When I `call close(6)` the process doesn't go into 100% cpu load any more. But it's hard to tell, if the process still does something at all.

So I used the original sml_server application. When I send the `call close(6)` to it, it returns with `libsml: error: unrecognized sequence`.

If this is the expected behaviour, then I need to add some recovery mechanism to my application, that automatically reopens the connection to the Zähler.

r00t-, 2018-10-31:
@tobylorenz:
this is expected.
the original code has no error handling at all, and will run into a loop as you observed.

the fix in #51 changes this in so far as it returns control to the caller.

due to lack of error reporting in the current api as it is,
it can only report the error by returning 0 from the read function, which can not happen in normal usage, and which your application has to handle.

re-opening the device internally does not seem feasible, as the handling is likely more complex, like sending an init sequence to the meter, which libsml knows nothing about.

tobylorenz, 2018-10-31:
Ok, then my assumption was right. The error handling has to be done in the application, for good reasons.
Good, from my side everything is fine then.
Let's merge the pull request and close this issue.

andig, 2018-11-02:
Fixed by #51

closed 2018-11-02 by andig

hmueller01, 2018-11-02:
Is the MQTT code somewhere online. I would be interested. Currently I am using sml_server and pipe the result to a Python script to send the data to a MQTT broker.

tobylorenz, 2018-11-04:
vzlogger has no support for MQTT. See https://github.com/volkszaehler/vzlogger/issues/308

My software is available here: https://bitbucket.org/tobylorenz/sml2mqtt.

andig, 2018-11-04:
Ups, I totally forgot. Another good alternative is nodered. You can use vzlogger‘s push functionality to push to nodered and relay to mqtt and other adapters.

hmueller01, 2018-11-04:
@tobylorenz Thanks a lot! I'll look into this soon.
@andig Yes, I know. No problem. :-) I'm using sml_server, but no vzlogger ...

mbehr1, 2018-11-04:
vzlogger mqtt support is here: https://github.com/volkszaehler/vzlogger/pull/357 

> 
> vzlogger has no support for MQTT. See volkszaehler/vzlogger#308 <https://github.com/volkszaehler/vzlogger/issues/308>
> My software is available here: https://bitbucket.org/tobylorenz/sml2mqtt <https://bitbucket.org/tobylorenz/sml2mqtt>.

Gruß

Matthias