volkszaehler/libsml #56: contributor agreement?

jmberg, 2018-12-12:
Since you're a fork from dailab, do you really want the contributor agreement still? Just curious.

andig, 2018-12-14:
Not really but I don't know if we are free to do so.

ping @justinotherguy

jmberg, 2018-12-14:
Of course you can remove it.

The original source code is GPLv3, so your only requirements are to adhere to that license. You don't need to actually assign copyright of your own contributions to them, it's yours!

I don't know what discussions you had on the mailing list you alluded to in the other ticket, but the situation here is really that clear.

jmberg, 2018-12-14:
For the record, I'm not a lawyer but an engineer, but I routinely deal with these things at my work - a large chunk of which consists of contributions to the GPLv2 licensed Linux kernel and similar projects.

The GPL (v2 or v3) has no requirements for copyright assignment, but if you want to license the library under a *different* license then you need to own all the copyright. That is clearly the intention of the original dailab code, since they write "other licenses are available".

I'd recommend that you:
 * remove the copyright assignment form (and any potential language referring to it)
 * remove the language saying "other licenses are available"
 * add your own copyright statements to the README.md file, unless you've actually sent copyright assignment forms to dailab?! this isn't really needed but it helps to be obvious

closed 2018-12-14 by jmberg