volkszaehler/libsml #58: example server should not use %.1f

jmberg, 2018-12-14:
The example server is printing all values with %.1f which loses precision if the scaler is less than -1, it should use %.*f instead. I've implemented that here:

https://github.com/jmberg/libsml/commit/31efb5f30eec23bcb33e2588d4cfa3e89a38ae54 ("sml_server: print values with the correct precision")

andig, 2018-12-14:
Please open a pr

jmberg, 2018-12-14:
You have a copyright assignment form which I have no intention of signing...

andig, 2018-12-14:
No problem. Obviously just taking your code won't fix that. Too bad we can't get contributions back in this case. If you want to contribute we'd be happy to welcome you on the developers mailing list to discuss the (inherited) copyright issue.

closed 2018-12-14 by andig

jmberg, 2018-12-14:
If you don't actually want the copyright .. I'll make a PR if/when you decide to remove it. FWIW, I've already removed it from my own repository, I'm not even pretending I'd assign any copyright to dailab for things I wrote.