volkszaehler/libsml #59: implement DZG meter workaround?

jmberg, 2018-12-14:
I found that my DZG DVS74 meter has a bug in the encoding of positive values in the range of 327.68..655.36W, it tags them as SIGNED but outputs them as 2 bytes in UNSIGNED format.

I don't know if the library or something else should have a workaround, but I've implemented one in the example server:
https://github.com/jmberg/libsml/commit/456d7e3ee7f36c58bed6468846f42ccd4067960c ("sml_server: implement DZG encoding bug workaround")

jmberg, 2018-12-17:
will send a PR once the ones I have now are decided upon

closed 2018-12-17 by jmberg

r00t-, 2023-07-05:
@jmberg:
do you have any code for this? (as this issue came up (again) in #125 )

jmberg, 2023-07-06:
The code is still there, just gated by a model/serial check. I'm travelling now but with the bug data I guess we can extend the workaround.