volkszaehler/libsml #68: Add note on README.md that this is the active libsml repo

devZer0, 2019-01-19:
google search for libsml still lists dailab repo as "the only one" and does not give a hint on volkszaehler/libsml  being the correct/active one ( see https://www.google.com/search?q=libsml )

so people could start taking a look at dead code and giving up frustrated

at least this repo should announce some info that it's a fork and that this is the active/maintained project and dailab repo is dead end.

maybe somebody with SEO experience could do something about the position in google search...

andig, 2019-01-19:
Why not open a pr instead of an issue?

devZer0, 2019-04-10:
sorry, i referenced the wrong one...that should have been https://github.com/volkszaehler/libsml/issues/67

devZer0, 2019-04-14:
this can be closed after merging the pull request

closed 2019-10-23 by andig