volkszaehler/libsml #69: warning: type of bit-field 'dzg_meter' is a GCC extension

devZer0, 2019-04-10:
src/sml_list.c:124:2: Warnung: Typ des Bitfeldes »dzg_meter« ist eine Erweiterung des GCC [-Wpedantic]
  u8 dzg_meter:1;
  ^
src/sml_list.c:124:2: warning: type of bit-field 'dzg_meter' is a GCC extension [-Wpedantic]
  u8 dzg_meter:1;
  ^

andig, 2019-04-11:
Should be a uint8 or simply uint?

devZer0, 2019-04-11:
i'm not good at c-programming, but isn't that struct simply meant so carry some flag to simply store some tiny bit of information if some workaround needs to be applied again or not?

don't know what correct datatype is correct for this, i would have chosen some "bool" but not some special type which generates some compiler warning.

so i think unsigned int would be ok here and makes the compile warning go away. 

should i send a pull request ?

devZer0, 2019-04-11:
by reading this, i thnk as you tell "unsigned int" is correct when using bit-field:

https://www.tutorialspoint.com/cprogramming/c_bit_fields.htm

devZer0, 2019-04-11:
so
```
struct workarounds {
        u8 dzg_meter:1;
};
```
should be
```

struct workarounds {
        unsigned int dzg_meter : 1;
};
```

devZer0, 2019-04-14:
this can be closed after merging the pull request

andig, 2019-09-22:
Bzw.: sollte das nicht lieber ein uint8_t sein (https://stackoverflow.com/questions/22842707/size-of-uint8-uint16-and-uint32#22843368) damit es wirklich 8-bittig bleibt? Oder ist uns das egal?

devZer0, 2020-02-21:
could you also close this one @andig ? (as i see that my fix proposal has been merged)

sorry, i did not see your question and i cannot answer, as my c-skills are not good enough

closed 2020-02-21 by andig