volkszaehler/libsml #75: [Holley DTZ541] Error while parsing SML message: Error: Wrong TL-Field 0x65 for Choice!

hacki11, 2019-09-22:
Hi,

unfortunately there is an issue with the Holley DTZ541 Smart Meter which is currently installed in broad area in germany.
Here are mentions of the issue:
https://github.com/Apollon77/smartmeter-obis/issues/63
https://github.com/coolchip/node-red-contrib-smartmeter/issues/8
https://forum.iobroker.net/topic/25091/holley-dtz541-sml-1-04/2

There are two problems:
1) SML_Time is wrong, missing bytes are:
    0x72: List of two
    0x62: uint8 
    0x01 (CHOICE secIndex)
    
```
Bytes in [] are not provided by smart meter
 77
    070100010800FF
    65001C0104
    [726201]65011C1EB7
    621E
    52FF
    6500DC5328
    01
77
    070100020800FF
    65001C0104
    [726201]65011C1EB7
    621E
    52FF
    6200
    01
```
Possible Fix: Introduce hard coded workaround and add bytes during parsing

2) CRC algorithm uses CRC-16/CCITT (KERMIT) instead of XMODEM
Possible Fix: Introduce a switch to choose correct crc (to be discussed)

I can provide a pull request for both issues but i think both solutions should be discussed before hand. Mostly the second issue, i am not sure how to recognize to choose the second crc algorithm. Trying both seems like a sledge hammer solution.

Additionally, i will inform my provider but the meter is directly from china and i do not think my provider will forward this information to the right people.

Last i have to say that its a absolute no go to roll out such a device which is NOT capable of SML. Because of both issues there is no available device to read out the values. Here are 1.8.0 and 2.8.0, two of the most important ones..

mbehr1, 2019-09-22:
libsml uses crc-16/ccitt, or?  where do you know that crc algo is wrong? (in any case I could imagine a autodetection e.g. on first package? if crc is wrong try other algo. if that's ok, stick with that (and retry until at least one correct crc is found and then stick to that algo)

hacki11, 2019-09-22:
there are several crc algorithms known as crc-16/ccitt.

SML uses
width=16 poly=0x1021 init=0xffff refin=true refout=true xorout=0xffff check=0x906e residue=0xf0b8 name="CRC-16/IBM-SDLC"

But Holleys DTZ541 uses
width=16 poly=0x1021 init=0x0000 refin=true refout=true xorout=0x0000 check=0x2189 residue=0x0000 name="CRC-16/KERMIT"

Where:
after the first message is parsed, the crc is calculated. At this point i could try another algo and set it for future usage. That could be a solution, yes.

I can only read the manufacturer, but i think making a decision on that is too broad. Perhaps they have some correct working meters.

hacki11, 2019-09-22:
there is no easy way to store some global information (store crc algo selection).
I now do the crc-workaround only if the original value mismatches.
That means the only negative effect will be in detecting a wrong crc a few eyeblinks later and a possible mismatch collision (mismatched crc but positive in kermit algo) but thats a really really rare corner case.

https://github.com/volkszaehler/libsml/pull/76

hacki11, 2019-09-22:
It seems the manufacturer already fixed the issue in devices from 2019

> https://www.loxforum.com/forum/projektforen/loxberry/plugins/85702-neues-plugin-smartmeter-stromzähler-auslesen?p=206977#post206977

magicWF, 2019-10-29:
Here in Wolfenbüttel starts the EVU to rollout the Holley DTZ-541. The calibration date is from 2019, but for decode the SML message the patch is require.

hacki11, 2019-10-29:
I can not understand how that can happen. Seems the EVU doesn’t even test the hardware they deploy. They will never be able to read them out with standard tools...

andig, 2019-10-29:
I don‘t think they want to. Plus- in due time- everything will be smart meters for that purpose...

Cheers,
Andi

Ps.: will merge the workaround unless there are any objections.

hacki11, 2019-10-31:
merged

closed 2019-10-31 by hacki11

andig, 2019-10-31:
Thank you for your patience!

spielkind1, 2022-05-25:
Weis jemand ob das auch bei Tasmota möglich ist? ich benutze das script von [hier ](https://tasmota.github.io/docs/Smart-Meter-Interface/#carlo-gavazzi-em340-modbus-rtu)welches natürlich wegen dem Fehler nicht funktioniert.

-------------------------------------------------
Does anyone know if this is also possible with Tasmota? I use the script from [here](https://tasmota.github.io/docs/Smart-Meter-Interface/#carlo-gavazzi-em340-modbus-rtu) which of course doesn't work because of the error.