volkszaehler/libsml #78: wrong results with EMH_eHZ-HW8E2A5L0EK2P_2.bin from testing repo

wbeck77, 2019-10-14:
Hi,

I added 3 files from two smart meters. One smart meter type is even new in your list.

The new one is parsed by libsml quite well, but the model you have already two other files makes problems:

cat ../../../Desktop/EMH_eHZ-HW8E2A5L0EK2P_2.bin | ./sml_server -
129-129:199.130.3255#EMH#
1-0:0.0.9255#06 45 4d 48 01 02 71 58 20 51 #
1-0:1.8.0255#13312484.9#Wh
1-0:1.8.1255#13312484.9#Wh
1-0:1.8.2255#0.0#Wh
1-0:15.7.0255#139.4#W
129-129:199.130.5*255#19 92 1a 52 65 64 ac ba e3 02 6e 61 35 26 75 ec 28 27 66 83 f0 2d bb 32 fa 70 1e 3a 75 3b 34 21 8b 64 66 b1 50 64 df 59 19 f9 f1 3f 84 b5 b2 c9 #

I do not know about the last output 129-129:199.130.5*255
Is this output correct? is the parser working well here? Maybe not.....

Hope my contribution helps a bit to make your great work even better!

Best regards

Wolfgang

devZer0, 2019-10-14:
the mentioned binary dump can be found here: 
https://github.com/devZer0/libsml-testing

devZer0, 2019-10-14:
i had a look at your problem.

you are right that the output may look weird from a first point of view, but i think it's quite correct - as obis-code  129-129:199.130.5*255 means "public key", which is a 96 digits number

see https://wiki.volkszaehler.org/software/obis

wbeck77, 2019-10-18:
crazy! Thanks for your investigation!
The data stream is 48 Bytes. So 96 nibbles
does "digit" mean "nibble" in your answer?

devZer0, 2019-10-18:
yes

devZer0, 2019-10-29:
can you please close this issue? thank you

closed 2019-10-29 by andig