volkszaehler/libsml #87: Not builing on current gcc version 10

freaknils, 2020-11-16:
libSML does not build fine on gcc 10.x

```
~% make clean && docker run --rm -v "$PWD":/usr/src/myapp -w /usr/src/myapp gcc:10 make   
make[1]: Verzeichnis „/home/nils/Downloads/libsml/sml“ wird betreten                                                                                                     
make[1]: Verzeichnis „/home/nils/Downloads/libsml/sml“ wird verlassen                                                                                                    
make[1]: Verzeichnis „/home/nils/Downloads/libsml/examples“ wird betreten                                                                                                
make[1]: Verzeichnis „/home/nils/Downloads/libsml/examples“ wird verlassen                                                                                               
make[1]: Verzeichnis „/home/nils/Downloads/libsml/test“ wird betreten                                                                                                    
make[1]: Verzeichnis „/home/nils/Downloads/libsml/test“ wird verlassen                                                                                                   
Unable to find image 'gcc:10' locally                                                                                                                                    
10: Pulling from library/gcc                                                                                                                                             
Digest: sha256:73b4afe8ee7c0d23b26ea5024dade9feec9d38eab2c144380078790e317585f5
Status: Downloaded newer image for gcc:10
make[1]: Entering directory '/usr/src/myapp/sml'
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_file.o src/sml_file.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_attention_response.o src/sml_attention_response.c                                                                                                                                                                                                  
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_transport.o src/sml_transport.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_octet_string.o src/sml_octet_string.c

src/sml_octet_string.c: In function 'sml_octet_string_parse':
src/sml_octet_string.c:80:17: warning: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'long unsigned int'} [-Wsign-compare]
   80 |  if (l < 0 || l >= (buf->buffer_len - buf->cursor)) { // sanity check: doesnt fit into buffer...
      |                 ^~
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_shared.o src/sml_shared.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_number.o src/sml_number.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_message.o src/sml_message.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_time.o src/sml_time.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_list.o src/sml_list.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_status.o src/sml_status.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_value.o src/sml_value.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_tree.o src/sml_tree.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_boolean.o src/sml_boolean.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_crc16.o src/sml_crc16.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_open_request.o src/sml_open_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_open_response.o src/sml_open_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_list_request.o src/sml_get_list_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_list_response.o src/sml_get_list_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_close_request.o src/sml_close_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_close_response.o src/sml_close_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_set_proc_parameter_request.o src/sml_set_proc_parameter_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_proc_parameter_response.o src/sml_get_proc_parameter_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_pack_response.o src/sml_get_profile_pack_response.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_list_request.o src/sml_get_profile_list_request.c
cc -I./include/ -fPIC -fno-stack-protector -g -std=c99 -Wall -Wextra -pedantic   -c -o src/sml_get_profile_list_response.o src/sml_get_profile_list_response.c
cc -Wl,-soname=libsml.so.1 -shared -o lib/libsml.so.1 src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src/sml_open_request.o src/sml_open_response.o src/sml_get_list_request.o src/sml_get_list_response.o src/sml_close_request.o src/sml_close_response.o src/sml_set_proc_parameter_request.o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_response.o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_response.o src/sml_get_profile_list_request.o src/sml_get_profile_list_response.o -luuid
ar -rs lib/libsml.a src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src/sml_open_request.o src/sml_open_response.o src/sml_get_list_request.o src/sml_get_list_response.o src/sml_close_request.o src/sml_close_response.o src/sml_set_proc_parameter_request.o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_response.o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_response.o src/sml_get_profile_list_request.o src/sml_get_profile_list_response.o
ar: creating lib/libsml.a
ld -r -o lib/libsml.o src/sml_file.o src/sml_attention_response.o src/sml_transport.o src/sml_octet_string.o src/sml_shared.o src/sml_number.o src/sml_message.o src/sml_time.o src/sml_list.o src/sml_status.o src/sml_value.o src/sml_tree.o src/sml_boolean.o src/sml_crc16.o src/sml_open_request.o src/sml_open_response.o src/sml_get_list_request.o src/sml_get_list_response.o src/sml_close_request.o src/sml_close_response.o src/sml_set_proc_parameter_request.o src/sml_get_proc_parameter_request.o src/sml_get_proc_parameter_response.o src/sml_get_profile_pack_request.o src/sml_get_profile_pack_response.o src/sml_get_profile_list_request.o src/sml_get_profile_list_response.o
make[1]: Leaving directory '/usr/src/myapp/sml'
make[1]: Entering directory '/usr/src/myapp/examples'
cc -I../sml/include/ -g -std=c99 -Wall -Wextra -pedantic -c sml_server.c -o sml_server.o
cc sml_server.o ../sml/lib/libsml.a -lm -luuid -o sml_server
make[1]: Leaving directory '/usr/src/myapp/examples'
make[1]: Entering directory '/usr/src/myapp/test'
make[2]: Entering directory '/usr/src/myapp/sml'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/usr/src/myapp/sml'
[... cut ...] sml_message_test.o ../sml/lib/libsml.a test_main.c -o test -luuid
libsml: error: message type 0900 not yet implemented
libsml: NYI: sml_message_body_free for message type 0900
libsml: warning: could not read the whole file
Unity test run 1 of 1
........................................................................................
-----------------------
88 Tests 0 Failures 0 Ignored
OK
make[1]: Leaving directory '/usr/src/myapp/test'
```

I think there maybe will be a simple solution?

Thanks!

devZer0, 2020-12-26:
which platform/linux/distro is using gcc10 yet?

you may have a look at https://wiki.gentoo.org/wiki/Gcc_10_porting_notes/fno_common  

please report if this helps

r00t-, 2020-12-27:
here's another project's fix to the same issue, with detailed comments:
https://github.com/plougher/squashfs-tools/commit/fe2f5da4b0f8994169c53e84b7cb8a0feefc97b5

r00t-, 2020-12-27:
@devZer0: 
> which platform/linux/distro is using gcc10 yet?

debian's next release will use it...

(copying my description from #90):
if this is not fixed, libsml (and dependant packages) won't be in the next debian release.

> Date: Fri, 17 Apr 2020 11:05:01 +0000
> The package fails to build in a test rebuild on at least amd64 with
> gcc-10/g++-10, but succeeds to build with gcc-9/g++-9. The
> severity of this report will be raised before the bullseye release,
> so nothing has to be done for the buster release.

> Severity set to 'serious' from 'normal'  (Wed, 22 Jul 2020 08:05:49 GMT)

-- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=957478

> The next release of Debian is codenamed bullseye — no release date has been set
> Debian 10 (buster) — current stable release

--- https://www.debian.org/releases/

the actual error seems rather trivial,
the tests fail to link due to conflicting variable names (`buf`). i remember reading something about changed variable scoping in object files.

```
make[2]: Leaving directory '/home/r00t/libsml/sml'
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -luuid unity/unity.o unity/unity_fixture.o src/test_helper.o src/sml_octet_string_test.o src/sml_buffer_test.o src/sml_number_test.o src/sml_boolean_test.o src/sml_value_test.o src/sml_status_test.o src/sml_list_test.o src/sml_time_test.o src/sml_tree_test.o src/sml_file_test.o src/sml_open_request_test.o src/sml_get_profile_pack_request_test.o src/sml_message_test.o ../sml/lib/libsml.a test_main.c -o test -luuid
/usr/bin/ld: src/sml_buffer_test.o:(.bss+0x8): multiple definition of `buf'; src/sml_octet_string_test.o:(.bss+0x8): first defined here
/usr/bin/ld: src/sml_number_test.o:(.bss+0x8): multiple definition of `buf'; src/sml_octet_string_test.o:(.bss+0x8): first defined here
/usr/bin/ld: src/sml_boolean_test.o:(.bss+0x8): multiple definition of `buf'; src/sml_octet_string_test.o:(.bss+0x8): first defined here
/usr/bin/ld: src/sml_value_test.o:(.bss+0x8): multiple definition of `buf'; src/sml_octet_string_test.o:(.bss+0x8): first defined here
```

r00t-, 2020-12-28:
there are two MRs to choose from, both work, the difference is somewhat esoteric.

* either explicitly make the variable a shared global, which is what the code did before
* or make them local to the defining file, which is/was the intuitive expectation.

r00t-, 2020-12-28:
@mbehr1, @andig:
maybe you want to comment check, as you are maintaining https://github.com/volkszaehler/vzlogger/commits/master/.travis.yml

r00t-, 2020-12-28:
i missed we already have https://github.com/volkszaehler/libsml/pull/88 by @jpfeiffer , should have just merged that :\

mbehr1, 2020-12-28:
yep, I'd prefer to merge #88 and not the new one with "extern". We could add the issue entry in the commit message?

closed 2020-12-28 by r00t-