volkszaehler/libsml #90: fix gcc-10 build

r00t-, 2020-12-27:
if this is not fixed, libsml (and dependant packages) won't be in the next debian release.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=957478

> Date: Fri, 17 Apr 2020 11:05:01 +0000
> The package fails to build in a test rebuild on at least amd64 with
> gcc-10/g++-10, but succeeds to build with gcc-9/g++-9. The
> severity of this report will be raised before the bullseye release,
> so nothing has to be done for the buster release.

> Severity set to 'serious' from 'normal'  (Wed, 22 Jul 2020 08:05:49 GMT)

> The next release of Debian is codenamed bullseye — no release date has been set
> Debian 10 (buster) — current stable release
https://www.debian.org/releases/

the actual error seems rather trivial,
the tests fail to link due to conflicting variable names (`buf`). i remember reading something about changed variable scoping in object files.

devZer0, 2020-12-27:
see https://github.com/volkszaehler/libsml/issues/87#issuecomment-751381454

devZer0, 2020-12-28:
this can be closed.

r00t-, 2020-12-29:
closing as a duplicate of #87 .
(thought i had done that two days ago)

closed 2020-12-29 by r00t-