volkszaehler/libsml #94: DZG - libsml: error: unrecognized sequence

mucheinz, 2021-01-27:
Hello,
when feeding data from a DZG meter to libsml e.g. using sml_server /dev/lesekopf0 the process exits after a couple of seconds with libsml: error: unrecognized sequence.
Reading through past issues it seems this is related to experiences of others with DZG meters.
I checked and adjusted the lesekopf at the meter but the situation is unchanged - seems there are some packets that libsml  does not interpret correctly.

1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225125.8#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#100.77#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225125.9#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#115.64#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225125.9#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#126.63#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225125.9#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#123.01#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.0#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#102.05#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.0#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#100.95#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.0#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#110.74#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.1#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#99.18#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.1#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#93.55#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.1#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#99.48#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.1#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#98.01#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.2#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#88.97#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.2#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#105.81#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.2#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#95.58#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.3#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#97.13#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.3#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#95.55#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.3#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#102.10#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.3#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#96.08#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.4#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#95.37#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.4#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#91.80#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.4#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#90.10#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.4#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#93.41#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.5#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#93.53#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.5#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#91.85#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.5#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#100.64#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.5#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#90.72#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.6#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#35.46#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.6#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#35.91#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.6#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#26.28#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.6#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#22.02#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.6#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#25.68#W
1-0:96.50.1*1#DZG#
1-0:96.1.0*255#0a 11 43 5a 57 01 02 81 4d 6f #
1-0:1.8.0*255#13225126.7#Wh
1-0:2.8.0*255#17498355.3#Wh
1-0:16.7.0*255#91.11#W
libsml: error: unrecognized sequence

devZer0, 2021-02-01:
can you provide a dump?  how to make one is described here. perhaps you should make a bigger one to include enough corrupted telegrams, so we perhaps can see a pattern...

https://github.com/devZer0/libsml-testing

mucheinz, 2021-02-01:
dump separately provided - four sample dumps - each containing corrupted telegrams.
I have a Lesekopf model "Udo" ;-) - connected via USB cable to a raspberry Pi. Have double checked connectivity and placement of the optical head a couple of times, re-set and re-booted Pi etc.

devZer0, 2021-02-03:
thanks. i will have a look

devZer0, 2021-02-03:
looks weird. in the second sml telegram of binary file DZG_DVS74-2.bin you sent to me, the entire stop escape sequence ( 1b 1b 1a 04 +fillbyte +checksum is missing (and so libsml is right telling unrecognized sequence). 

just before the next telegram i see lot's of "ff" and bytes with higher values, so i guess this is optical or electrical interference or problems with the serial communication. 

so i think this is no parser error but you get scrambled data from your smartmeter.

not sure how to solve...

- check serial comm params
- make sure the opical probe is shielded from environmental light
- try with a different client/pc (notebook)
- try optical probe + raspi with a different smartmeter

to get a clue what's going on

`0000000: 1b1b 1b1b 0101 0101 7605 e0ef 670a 6200  ........v...g.b.
0000010: 6200 7263 0101 7601 0102 310b 0a01 445a  b.rc..v...1...DZ
0000020: 4700 0281 4d6b 7262 0165 05a7 c454 6202  G...Mkrb.e...Tb.
0000030: 6376 8000 7605 e1ef 670a 6200 6200 7263  cv..v...g.b.b.rc
0000040: 0701 7701 0b0a 0144 5a47 0002 814d 6b07  ..w....DZG...Mk.
0000050: 0100 620a ffff 7262 0165 05a7 c454 7577  ..b...rb.e...Tuw
0000060: 0701 0060 3201 0101 7262 0162 0062 0052  ...`2...rb.b.b.R
0000070: 0004 445a 4701 7707 0100 6001 00ff 0172  ..DZG.w...`....r
0000080: 6201 6200 6200 5200 0b0a 0144 5a47 0002  b.b.b.R....DZG..
0000090: 814d 6b01 7707 0100 0108 00ff 641c 0104  .Mk.w.......d...
00000a0: 7262 0162 0062 1e52 ff65 07ed eb10 0177  rb.b.b.R.e.....w
00000b0: 0701 0002 0800 ff01 7262 0162 0062 1e52  ........rb.b.b.R
00000c0: ff65 0a6e cbc3 0177 0701 0010 0700 ff01  .e.n...w........
00000d0: 7262 0162 0062 1b52 fe54 0197 7a01 0101  rb.b.b.R.T..z...
00000e0: 63a6 da00 7605 e2ef 670a 6200 6200 7263  c...v...g.b.b.rc
00000f0: 0201 7101 6379 4900 0000 0000 1b1b 1b1b  ..q.cyI.........
0000100: 1a04 06c0 1b1b 1b1b 0101 0101 7605 e3ef  ............v...
0000110: 670a 6200 6200 7263 0101 7601 0102 310b  g.b.b.rc..v...1.
0000120: 0a01 445a 4700 0281 4d6b 7262 0165 05a7  ..DZG...Mkrb.e..
0000130: c455 6202 63e3 2d00 7605 e4ef 670a 6200  .Ub.c.-.v...g.b.
0000140: 6200 7263 0701 7701 0b0a 0144 5a47 0002  b.rc..w....DZG..
0000150: 814d 6b07 0100 620a ffff 7262 0165 05a7  .Mk...b...rb.e..
0000160: c455 7577 0701 0060 3201 0101 7262 0162  .Uuw...`2...rb.b
0000170: 0062 0052 0004 445a 4701 7707 0100 6001  .b.R..DZG.w...`.
0000180: 00ff 0172 6201 6200 6200 5200 0b0a 0144  ...rb.b.b.R....D
0000190: 0411 0240 516c 40b7 0000 0121 003f 6428  ...@Ql@....!.?d(
00001a0: 0070 6000 6262 1842 f661 066d 8203 3607  .p`.bb.B.a.m..6.
00001b0: 0002 0800 ff01 130b 4200 621e 00fe 4100  ........B.b...A.
00001c0: 06c3 0401 6607 0100 0007 00ff 0070 00c0  ....f........p..
00001d0: 6c22 1242 fc20 0081 4a01 0002 411e 1100  l".B. ..J...A...
00001e0: 7200 459d 0001 4820 0050 c000 0044 0064  r.E...H .P...D.d
00001f0: 3046 0000 0000 1216 1102 6130 73df fffe  0F........a0s...
0000200: 9f3f fcfe dfff ffbf 83c3 dfff ffff ffb3  .?..............
0000210: f7fe 7cff b03e ffdf fe7c 7ff6 f3fe ffec  ..|..>...|......
0000220: 8cff febf b0ee 6fb7 cffb e3fb cfff e3bf  ......o.........
0000230: 37fb e3b8 bcbb ecfc ecfc 1b1b 1b1b 0101  7...............
0000240: 0101`

devZer0, 2021-02-08:
did you receive the answer? any progress with this issue?

andig, 2021-02-18:
Ping @mucheinz

mucheinz, 2021-02-20:
Thank you for investigating, will continue researching on my side as well (have re-positioned the optical reader (replaced it with another), checked the cable etc... will come back if I have further details and close the issue for now.

closed 2021-02-20 by mucheinz

r00t-, 2023-01-22:
likely related to https://github.com/volkszaehler/libsml/issues/103