volkszaehler/libsml #95: More Examples

smart4energy, 2021-02-18:
I started working with libsml. So i try to understand this library. 

The server example is good. But I miss an example for creating a SML Message to request values from an bidrectional meter.

An perhaps a short example to create a client Example. Creating list_response_message.

So could someone who are familar with sml can create such examples ? Or help howto create these Messages.

So I can test the request/response on my machine with the Server and a Client.

devZer0, 2021-02-18:
what type of meter do you use? i think you are talking about what's being called "pullseq" in vzlogger. i'm not sure if libsml is being used for that, at least i did not see it yet.

r00t-, 2021-02-18:
sml is not a bi-directional protocol, if a meter accepts input (like to trigger sending data), that input is not in sml format.

and the only senders of sml are meters, and receivers are software, i don't think sml even implements sending, as there's no use-case for it.
(while from a purely academic point of view it might be useful for testing, we could only ever test against ourselves anyway.)

smart4energy, 2021-02-18:
SML is also a bidrectional protocol, because we have an requests Message and response Message in this protocol. So it also can used for this Case. 
In Germany a Meter Like ehz has a ir Interface for customer and in the back a ir Interface for the meter operator. And im interested to read from the back interface because of Security.
The protocol is bidrectional sml protocol. 
Also the bidrectional is described in German Papiers about this protocol.

Open requests Message 
Messages to request for example get list request
Close request Message 

The answor would

Open response Messag
Get list Response // this will be Periodic Pushed by the Front Interface. But then it is called  get list Response without reques.
Close Response Messag

So i have to create These Messages with this library i think it is possible ?

andig, 2021-02-18:
You‘re welcome to join the mailing list for open discussions. An issue tracker is likely not the best place for doing so.

antonmeyer, 2023-03-19:
here is an example of sending SML message.
https://github.com/dom3141/LMN_Communication_Example/blob/master/sources/sml.cpp
at least I got some inspiration
I want to disable the PIN via the MSB interface (MSB = MessStellenBetreiber)

@andig what mailing list you refer to?

andig, 2023-03-19:
See https://wiki.volkszaehler.org/contact