volkszaehler/libsml #98: Compile Error on Alpine: no member named '__jmpbuf'

m-reuter, 2021-03-25:
Hi, 
thank you for maintaining this project! I am getting a compile error on Alpine: 

```
cc -I../sml/include/ -std=c99 -Wall -Wextra -pedantic -c unity/unity.c -o unity/unity.o
unity/unity.c:23:21: error: 'struct __jmp_buf_tag' has no member named '__jmpbuf'
   23 |   .AbortFrame = { {.__jmpbuf = {0}, .__mask_was_saved = 0, .__saved_mask = {{0}} } }
      |                     ^~~~~~~~
unity/unity.c:23:38: error: 'struct __jmp_buf_tag' has no member named '__mask_was_saved'
   23 |   .AbortFrame = { {.__jmpbuf = {0}, .__mask_was_saved = 0, .__saved_mask = {{0}} } }
      |                                      ^~~~~~~~~~~~~~~~
unity/unity.c:23:61: error: 'struct __jmp_buf_tag' has no member named '__saved_mask'
   23 |   .AbortFrame = { {.__jmpbuf = {0}, .__mask_was_saved = 0, .__saved_mask = {{0}} } }
      |                                                             ^~~~~~~~~~~~
unity/unity.c:23:3: warning: braces around scalar initializer
   23 |   .AbortFrame = { {.__jmpbuf = {0}, .__mask_was_saved = 0, .__saved_mask = {{0}} } }
      |   ^
unity/unity.c:23:3: note: (near initialization for 'Unity.AbortFrame[0].__ss[0]')
make[1]: *** [Makefile:40: unity/unity.o] Error 1
make[1]: Leaving directory '/tmp/libsml/test'
make: *** [Makefile:5: all] Error 2
```

Replicate: 
````
docker run -it --rm alpine /bin/ash
apk add --no-cache  build-base git util-linux-dev
git clone https://github.com/volkszaehler/libsml.git
cd libsml
make
```

Thanks for any pointers...

r00t-, 2021-03-25:
unity is the test-framework being used,
and it would be a bug on their end.
(unless we use an outdated version and they fixed it already).
simply run `make -C sml` instead of `make` to skip the tests.

m-reuter, 2021-03-25:
thanks, I noticed also simply using:
`struct _Unity Unity = { 0 , 0 , 0 , 0 , 0 , 0 , 0 , 0 , { 0 } };  `
from the ifdef __APPLE__ block instead of the other one seems to fix this. 
Not sure how these lines make it into the unity.c file from the Unity project that you linked? And not sure how to report it there as I am not familiar with this framework. 
One problem is, that the make command is called from vzlogger and I don't think there is an option there to skip the test?

How should this be resolved?
- modify the lines in unity.c to always use the APPLE block (what is the other one used for anyway)?
- check whether the unify project above has a fix already, if not find out how to report it there?
- add a flag to vzlogger to skip tests?

Thanks

andig, 2021-03-25:
Should be fixed upstream, not here

m-reuter, 2021-03-25:
Don't think it is necessarily an upstream issue. See here #41 where these lines were added.

devZer0, 2021-03-25:
yes, it isn't, that piece of code does not exist upstream

r00t-, 2021-03-25:
we DO use an outdated version, last update january **2013**...
https://github.com/volkszaehler/libsml/commits/master/test/unity

we should try simply copying the files from https://github.com/ThrowTheSwitch/Unity/tree/master/src
and seeing if that works better.
should we maybe make that a git submodule to avoid the copying?
(but that would probably make the build process too complicated for some users.)

r00t-, 2021-03-25:
@m-reuter:
vzlogger will surely not invoke `make`...
`install.sh` does. https://github.com/volkszaehler/vzlogger/blob/master/install.sh#L226
imho that script is an abomination that should not exist ;)
but we might make it only warn but not abort if `make test` fails.

r00t-, 2021-03-25:
simply copying the files from unity does not work, their API has changed sice 2013 ;)
somebody will have to look into this in more depth.

m-reuter, 2021-03-25:
@r00t-  yes, I also would love to see install.sh be replaced by a nice cmake approach. 
Modifying install.sh can either: i) always skip the test, ii) skip the test if a flag is passed, iii) separate building the library and testing and continue if the test fails. 
Anyway I think it would be easier to find out why these changes were introduced in this project in #41 and not simply splitting in __APPLE__ and rest of world, but maybe only using the problem line for the OSs where this is needed (raspbian and what else?). Or we just add Alpine to the __APPLE__ block.

r00t-, 2021-03-25:
@m-reuter:
"install.sh" does not replace cmake.
it replaced something like a package builder or whatever, building the project AND it's dependencies for users not capable of doing so themselvs.

m-reuter, 2021-03-25:
cmake can build dependencies and can also install the package. And for the users who do not want to type the cmake command, there could still be a simplified install.sh. But this is really an issue that should be discussed at vzlogger. 

For the issue here I think the fastest solution is to fine-tune #41 to be more specific about what OS should use what line. Adopting the new unify api is probably more involved (but probably necessary in the long run to maintain this project to work with more up-to-date dependencies).  Maybe @andig can comment as he introduced #41 ?

r00t-, 2021-03-25:
#99 builds on your weird distro:
```
$ docker run -it --rm alpine /bin/ash -c 'apk add --no-cache build-base git util-linux-dev && git clone -b upgrade_unity_to_v2.5.2 https://github.com/r00t-/libsml.git && cd libsml && make'
[...]
88 Tests 0 Failures 0 Ignored 
OK
make[1]: Leaving directory '/libsml/test'
```

m-reuter, 2021-03-25:
Thanks, will take a look. 
While I am also not an alpine fan, it is gaining a lot of importance as a base docker image since it is so small. Also large frameworks (e.g. home assistant) use it for their Add-ons.

closed 2021-03-29 by r00t-