volkszaehler/libsml #101: Unsigned current Power Value

SirkoVZ, 2021-03-29:
According to the SML protocol
https://www.vde.com/resource/blob/951000/252eb3cdf1c7f6cdea10847be399da0d/fnn-lastenheft-edl-1-0-2010-01-13-data.pdf
page 31 the direction (consumption or delivery) is marked using flag.
It seems the SMLlib is not honoring the flag, as it always shows positive values, even when delivering power.
At least in the EMZ ED300L no negative values are shown when delivering power
The (un)sign seems to be addressed already in Tasmota:
https://github.com/arendst/Tasmota/issues/8001#issuecomment-604868066
and finally in
https://github.com/arendst/Tasmota/pull/8020/commits/61b97350daefe4581162994b38482d241f7ecfcd
Maybe we can also add this in the SML lib...

r00t-, 2021-03-29:
can you attach or link binary dumps of sml transmissions with both variants?

SirkoVZ, 2021-03-29:
How do I do that?
dd if=/dev/ttyAMA0 of=/tmp/dumpfile.bin bs=1 count=4096
doesn't create any file, while vzlogger works properly using /dev/ttyAMA0

r00t-, 2021-03-29:
@SirkoVZ:
in theory that should work,
bute note that using `dd` os never necessary, and kind of silly on a character device.
just `cat /dev/ttyACM0 >logfile`
if in doubt also try `xxd </dev/ttyACM0` (a hexdump also works, as it can be converted back. and actually it can simply be posted here as text.)

r00t-, 2021-03-29:
oh, look, it's all here:
https://github.com/devZer0/libsml-testing
you could just submit your dumps right there

r00t-, 2021-03-31:
yay, data!

first results look good:
i added this:
```
[~/libsml]$ git diff examples/
diff --git a/examples/sml_server.c b/examples/sml_server.c
index 1de47bc..83a775b 100644
--- a/examples/sml_server.c
+++ b/examples/sml_server.c
@@ -131,6 +131,12 @@ void transport_receiver(unsigned char *buffer, size_t buffer_len) {
                                        if (prec < 0)
                                                prec = 0;
                                        value = value * pow(10, scaler);
+                                       if (entry->status)
+                                               printf(
+                                                       "status: %x (type %x)\n",
+                                                       *(entry->status->data.status16),
+                                                       entry->status->type & SML_TYPE_FIELD
+                                               );
                                        printf("%d-%d:%d.%d.%d*%d#%.*f#",
                                                entry->obj_name->str[0], entry->obj_name->str[1],
                                                entry->obj_name->str[2], entry->obj_name->str[3],
```

and get:
```
[~/libsml]$ ./examples/sml_server -v - <../libsml-testing/EMH-ED300L_consumption.bin | grep -A1 status
status: 182 (type 60)
1-0:1.8.0*255#17243368.3#Wh
[~/libsml]$ ./examples/sml_server -v - <../libsml-testing/EMH-ED300L_delivery.bin  | grep -A1 status
status: 1b0 (type 60)
1-0:1.8.0*255#17243675.6#Wh
--
status: 1b0 (type 60)
1-0:1.8.0*255#17243675.6#Wh
```
consumption = 0x182 = 0b1100**0**0010
delivery: 0x1B0 = 0b1101**1**0000

according to the docs
https://www.vde.com/resource/blob/951000/252eb3cdf1c7f6cdea10847be399da0d/fnn-lastenheft-edl-1-0-2010-01-13-data.pdf

> bit 5 Energierichtung 0 <=> +A; 1 <=> -A. 

so indeed in the delivery case, bit 5 is `1`!

the code in tasmota is also checking bit 5 (0x08):
https://github.com/arendst/Tasmota/commit/61b97350daefe4581162994b38482d241f7ecfcd#diff-97c9ae02f894436fc4188ce195998aa3d258fed1e78d21239c11c0c8539d01e5R944

r00t-, 2021-03-31:
@SirkoVZ:
are you using vzlogger or some other utility using libsml?

i'm not sure if this should be implemented in libsml or in it's user.
it seems to me that at the sml level, libsml extracts the data just fine,
the _meaning_ of the bits in the status field is not defined by sml, itself,
but in that vde document.
(just like the meaning of the obis numbers i guess)
so i don't think there's any need to change anything in libsml itself.

r00t-, 2021-03-31:
https://github.com/volkszaehler/vzlogger/pull/476 has an experimental implementation for vzlogger that works with the data in your dumps

SirkoVZ, 2021-03-31:
I can confirm that it is working... more details later,

thanks so far :-)

r00t-, 2021-04-02:
closing this, as it's out of the scope of libsml, see https://github.com/volkszaehler/vzlogger/issues/431

closed 2021-04-02 by r00t-