volkszaehler/libsml #102: timeout

ckohrt, 2021-11-09:
Hi!
I use the sml_server for getting the counter values with the command:
`Z1_280=$(/usr/bin/sml_server /dev/zaehler_1 | head -n10 | grep -oP '2\.8\.0\*255#\K.*(?=#)')`

This call is blocking.

I have 4 counters and if one of the IR reading heads does not answer, it blocks forever. So, it would be fine to have a timeout value.

At the moment I solved it with this:
`
while true; do
dd if=/dev/zaehler_1 of=Z1.bin bs=1 count=8192 iflag=fullblock status=none &
dd if=/dev/zaehler_2 of=Z2.bin bs=1 count=8192 iflag=fullblock status=none &
sleep 30
pkill dd
`
Two counters for getting the values Z1 1.8.0/2.8.0 and Z2 1.8.1/1.8.2/2.8.0.

Unfortunately I have no idea how to integrate a timeout into the sml_server.

cheers
Christian

PS
Nice work!

andig, 2021-11-09:
/cc @r00t-

r00t-, 2021-11-09:
@ckohrt:
i'm not sure if this within the scope of libsml...

`sml_server` is a simple program intended to demonstrate the use of the library (it lives in an "examples" directory).
`sml_server` uses some simple i/o utility functions that libsml provides in `sml_trasport.[ch]`,
which use blocking i/o and do not support timeouts.
i'd say, you are very welcome to write your own program using libsml that implements more advanced semantics.
by replacing `sml_transport_listen()` and `sml_transport_read()` with your own code, and passing the data to `sml_transport_receiver()`.
we _might_ want to offer a more flexible variant of sml_transport_read() that can be used with non-blocking io, or otherwise provide the scanning for start sequence and reading of a datagram in a more generic interface.
but as stated above, now sure if this is within scope,
it would also mean an ABI change.

why are you not using vzlogger?
it can read multiple meters in a single process and forward the data to your processing of choice.

as an alternative, if you insist on scripting around `sml_server`, have you considered simply wrapping sml_server in https://www.gnu.org/software/coreutils/timeout ?

ckohrt, 2021-11-10:
hi @r00t-!
Thank you very much. I ended up with:

`Z_1_182=$(timeout --foreground 10 /usr/bin/sml_server /dev/zaehler_1 | head -n10 | grep -oP '1\.8\.2\*255#\K.*(?=#)')`

which works. Of course vzlogger is a better alternative, but I didn't want to get vzlogger running on my raspberry as I already have a working solution. It's just the time invest.... and with the timeout solution it works nearly without any time invest!

Big thanks!
Christian

closed 2021-11-10 by ckohrt

r00t-, 2021-11-11:
@ckohrt:
`--foreground` should not be necessary.

ckohrt, 2021-11-11:
> @ckohrt: `--foreground` should not be necessary.

Thanks!

ckohrt, 2021-11-12:
Solution:
`Z_1_182=$(timeout 10 /usr/bin/sml_server /dev/zaehler_1 | head -n10 | grep -oP '1\.8\.2\*255#\K.*(?=#)')`