volkszaehler/libsml #110: SML-Errors with Holley DTZ541

oeiber, 2022-11-23:
Hi everybody,
I have a Holley DTZ541 which is connected to my RaspberryPi using a rs485 to serial converter from weidmann elektronik.
In the vzlogger I can see the following log entries. They occour multiple times per hour.

`[Nov 23 01:05:52][sml]  reopen called. current fd=4`

In stdout i can also see these entries multiple times per hour:

`libsml: error: unrecognized sequence`

A few weeks ago I was using an infrared head which was connected to my meter. I saw also the log entries from above. I thaught this is related to the weak shining ir led built into the meter. 
But after changing to a wired connection, I think the problem must be elsewhere. I do not think the meter is sending fragmentated messages periodically over a half duplex wired connection.

Maybe it is related to the libsml?

Thank you for your help!

matthiasklein, 2022-11-24:
Hello @oeiber,
I also regularly experience such problems with a Holley DTZ541-BDBA (year of manufacture 2021) in conjunction with an infrared head. I have no access to the RS485 because the meter is sealed.

Can you create a dump with dd over a longer time as described here?
https://github.com/devZer0/libsml-testing

Maybe someone can see something in there.

I have to omit the "iflag=fullblock" option on mine to write usable dumps.

r00t-, 2023-02-09:
quoting @fkohlgruber from https://github.com/devZer0/libsml-testing/pull/22#issuecomment-1340109611

> The data looks weird. It's quite a large dump, yet there's no single transmission end marker (1B1B1B1B1AXXYYZZ) in the it. When searching for 1B1B1B1B (which is 3 times 1B, not 4 times like it should be), there are 182 matches and most of them look like (incomplete) transmission start / end markers. To me it looks like the data is incomplete and bytes are missing everywhere. Are you sure the data collection works correctly?

closing this as invalid or duplicate of https://github.com/volkszaehler/libsml/issues/103

closed 2023-02-09 by r00t-