volkszaehler/libsml #113: hexdump() compatibility issue with newest Arduino / ESP8266 libraries

mh-er, 2023-01-27:
There is a name collision of function hexdump() in libSML/sml/src/sml_shared.c and libSML/sml/include/sml_shared.h with the newest Arduino / ESP8266 libraries.

Proposal: rename to sml_hexdump()

r00t-, 2023-02-09:
https://github.com/volkszaehler/libsml/blob/74e18fccc15bbd881a59befae1f6222f7047921d/sml/src/sml_shared.c#L165

r00t-, 2023-02-09:
https://github.com/volkszaehler/libsml/blob/74e18fccc15bbd881a59befae1f6222f7047921d/sml/include/sml/sml_shared.h#L110

i wonder if this is part of our public API and if we break any compatibility by removing it?

but a function with such a generic name should not have been exported anyway, and hopefully nobody ever used it.

narc-Ontakac2, 2023-06-10:
If a version bump is desired - I'll do a release as a basis for a debian package release soon. I'd suggest I'll go to 1.1.0 instead of 1.0.2 and you merge this PR before.

mh-er, 2023-07-31:
Saw your message just now. I re-based to the latest master. So you could take over the PR.

narc-Ontakac2, 2023-07-31:
Unfortunately I am the packager. All I do in this project is maintaining the debian packaging. So +1 for me for merging this, but I am not the one to do it.

r00t-, 2023-08-01:
@narc-Ontakac2:
please do the version bump, i'll merge

narc-Ontakac2, 2023-08-01:
The merge comes first. I'll then do a release 1.1.0 and from that a Debian release.

narc-Ontakac2, 2023-08-02:
Since @r00t- decided to merge I did it. Release 1.1.0 will follow.

closed 2023-08-02 by narc-Ontakac2