volkszaehler/libsml #115: libsml debian package still points to dailab repo

devZer0, 2023-02-13:
see 

https://packages.debian.org/source/sid/libsml
https://packages.debian.org/source/buster/libsml
https://packages.debian.org/source/bullseye/libsml
https://packages.debian.org/source/bookworm/libsml

-> External Resources:

->    [Homepage](https://github.com/dailab/libsml) [github.com]

who is able to resolve this?

also see
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1028076

and
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=946525

r00t-, 2023-02-14:
see also the oooold thread at https://github.com/volkszaehler/libsml/issues/72

@narc-Ontakac2 is already taking care oft this, see https://github.com/volkszaehler/vzlogger/issues/558

we're mostly waiting for https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1028076
to recevive no response for long enough to allow the package to be considered orphaned.

narc-Ontakac2, 2023-05-13:
I have in between triggered the package salvaging process with 
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1034250
and Andreas Moog passed me maintainership. As soon as the release freeze is over I will do an upload to unstable (sid) from where it will go to testing (trixie) . The existing bugs do however not justify changing the by then released distros.

narc-Ontakac2, 2023-07-11:
The Debian release 1.0.2-1 of libsml fixes this. It has been uploaded to sid.

closed 2023-07-11 by narc-Ontakac2