volkszaehler/libsml #131: sml_server not part of libsml1 debian package

devZer0, 2023-08-16:
any reason why libsml1 package only provides libsml library but , no binary build of sml_server is being provided as part of the package or an extra package ?

i needed that on my old raspberry today and had a hard time to get it built there

that binary is extremely useful for testing smart meters or explain sml with test binaries from libsml-testing repo

narc-Ontakac2, 2024-01-02:
Since I am mainly interested in vzlogger I wasn't  even aware it exists.

devZer0, 2024-01-02:
you can use it to directly convert serial data stream into readable values

you can feed data via stdin ( https://github.com/devZer0/libsml-testing ) or open serial device directly (https://community.openhab.org/t/using-a-power-meter-sml-with-openhab/21923 )

i think it's production quality, using it for years @dingfabrik cologne energy consumtion gauge

narc-Ontakac2, 2024-01-02:
I'll check this. If it is added to the package there should a man page. I'll see if I can write one. If it is added to the official Debian package a man page is a must.

narc-Ontakac2, 2024-01-03:
# NAME

sml_server - makes libsml functionality available on the command line

# SYNOPSIS

**sml_server** [**-v**] [**-h**] [**-s**] _device_

# DESCRIPTION

The sml_server reads binary SML data from the given device. If - is given, 
stdin is used. This data is then displayed in human readable form.

**-v**
:     This enables verbose output.

**-h**
:     This shows the help.

**-s**
:     What does this do?

# COPYRIGHT
The sml_server program is Copyright (C) Juri Glass, Mathias Runge, Nadim El 
Sayed and others.

# SEE ALSO
[libsml-testing](https://github.com/devZer0/libsml-testing)

narc-Ontakac2, 2024-01-03:
This is a suggestion for a minimal man page. Please make useful additions if possible and gie an explanation for -s.

r00t-, 2024-01-03:
`-s` will parse the first sml message(?) and then exit:
https://github.com/volkszaehler/libsml/blob/master/examples/sml_server.c#L147-148
the comment incorrectly calls this an "obis data stream", i'm pretty sure that's not valid sml terminology:
https://github.com/volkszaehler/libsml/blame/0051da9e439dcb2c80df4736008c2f9be6e76460/examples/sml_server.c#L42

it should be "device or file".

i would also mention that it's only a test/example program and not intended for "production" use.

closed 2024-01-06 by narc-Ontakac2

narc-Ontakac2, 2024-01-06:
Unfortunately a release is currently not possible due to a missing dependency in the Ubuntu repository.

PR #137 (2024-01-06): Added a small Debian package for sml_server
This closes #131. I'll wait a week for reviews and complaints. Then I'll merge and do a release.