volkszaehler/libsml #146: How to use libsml in ESPhome?

Qhilm, 2024-02-20:
Even though ESPhome now has [SML support](# https://github.com/esphome/esphome-docs/blob/current/components/sml.rst), it does not have a specific [workaround](https://github.com/volkszaehler/libsml/pull/136) I need for my WS7412.1.

Hence my question: can I replace this ESPhome yaml (for the ESPHome built-in SML support):

```yaml
external_components:
  - source:
      type: local
      path: components
    components: [ sml ]
```

with

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/volkszaehler/libsml
    components: [ sml ]
 ```
 
 as a drop in replacement? Or is it more complex than this?
 
I intend to test this, but I wanted to hear if someone can say I am completely on the wrong track here and it will not work at all like that, I can try to reproduce the workaround either in ESPhome or Home Assistant yaml while using the built-in SML support from ESPHome.

This would be with ESPhome 2023.12.9 and Home Assistant 2024.01.1.

r00t-, 2024-03-07:
i am aware that libsml should be usable on esp32,
but afaik nobody here does that, and even less using "ESPhome" or "Home Assistant".

they seem to have their own sml implementation:
https://github.com/esphome/esphome/blob/dev/esphome/components/sml/sml.cpp

so some more work would be needed to include libsml there.

i suggest you ask at "ESPhome" instead, asking if they can include libsml, or rather you can just look at their code yourself and see if you can adapt it to use libsml internally.

r00t-, 2024-03-07:
a web search says that somebody adapted libsml for arduino, which should then build for esp, you could start with that:
https://forum.arduino.cc/t/libsml-parsing-smart-meter-language-protocol/1149684

closed 2024-03-07 by Qhilm